betaflight/betaflight-configurator #1640: Make updating configuration more pleasant

atomgomba, 2019-09-10:
There has been discussion about making the transfer of settings across crafts easier. I was thinking of ways to do this earlier myself as I like to keep my non-specific settings consistent, ie. same OSD, same rates, same modes, etc. on all of my crafts and it is a little bit time consuming to connect, switch to CLI tab, click the button, select the file and so forth. This proposal is a potential solution within the Configurator as currently one can also do something like piping config files from stdin into a serial terminal, but it is not trivial or comfortable to do on some systems.

This is a rough mockup of the idea (I know it looks awful at the moment):

[image]

A new subtab in the firmware flasher could serve two use cases:

* User flashes new firmware with chip erase and execute existing diff/dump upon success in one go
* User wants to apply the same configuration to multiple crafts and this way she only has to select the file once, connect the craft, click Apply, connect another craft, click Apply, connect another craft, click, ...

I'm looking for comments on this idea.

atomgomba, 2019-09-10:
The "Ask to apply after flashing" switch would show the user a confirmation dialog whether she wants to apply the selected config file after successful flashing or not. This new tab later could be extended to include community presets pulled from some network source or maybe custom URL.

quadwizard, 2019-09-13:
This is an excellent concept and one I have been manually managing in my file system and CLI for as long as I have had quads :(

If this can be pulled together it would be awesome!

Mateyhv, 2019-09-24:
Good idea, definitely supporting it!

haslinghuis, 2021-06-05:
@limonspb is working on presets

closed 2021-06-05 by haslinghuis