betaflight/betaflight-configurator #1655: Build #306 does not flash firmware

cusglider, 2019-09-13:
Although visibly in DFU mode, the build #306 does not flash firmware. I had to go back to vers 10.51 to achieve this.

Secondly if you work inside the PID page it sometimes happens, that you cannot leave it, only by closing and restarting the application.

IvoFPV, 2019-09-14:
@cusglider Could you please provide more information on what do you mean by "does not flash firmware". Does it go through the flashing process in the status bar?
On the PID tab lock up, could you give me some steps to reproduce the issue? Also, could you please get the latest build you check next time?

cusglider, 2019-09-14:
Version #306 was the one available to flash the RC2 of betaflight 4.1 as of yesterday. I could connect to the quad alright, select the fc and desired firmware version, download it, however, after I pressed the "flash" nothing happened. I tried this on a Matek F405 and Matek F722 SE with the same result. 

As for the PID lock up I just looked into various values and filter settings and got stuck when I wanted to go to the next page. I did not count how many values I pointed my mouse to or entered different values.

IvoFPV, 2019-09-14:
@cusglider I can't reproduce any of the issues you've found. Flashing works as expected, pid tab works just fine.
>however, after I pressed the "flash" nothing happened

When pressing flash firmware did your FC reboot into DFU mode? Maybe a driver issue there.

>As for the PID lock up I just looked into various values and filter settings and got stuck when I wanted to go to the next page. I did not count how many values I pointed my mouse to or entered different values.

It is hard to track an issue down if it isn't repeatable and you can't provide step by step how to make it happen. It would be great if you could try to replicate the issue and find out what stops it from working, that's the only way that will get fixed.

cusglider, 2019-09-14:
Thanks for following up so closely.

I can only make you a video showing the steps, if that helps. The FC was on DFU mode as indicated by the configurator and it worked perfectly with the same FC, the same setting, the same firmware to be flashed on 10.51. So I have no indications of a driver issue. It all happened on a windows notebook.

IvoFPV, 2019-09-14:
@cusglider That would help a lot. I assume it is something to do with your configurator installation then.

cusglider, 2019-09-14:
Im having a bright smile as I write these lines ;-) . Just got aware of having installed when I returned from the airfield tonite the latest build #310 before I cared to check my mail. As I went thru the downloading and flashing process with that build couple minutes back and recorded all this I discovered that  n o w  it works. Thanks for your efforts and thanks for your dedication to the fpv community

tarite, 2019-09-18:
I had the same problem with build #310, updated to #311 with same results.
Spent way too much time messing with driver reinstalls, zadig and impulse fixer.
Removed chrome version and installed the windows 10.6 rc1.
Flashing is working now.
FC is the Iflight Succex TwinG 
This is from the config tool log using #311:
2019-09-14 @ 15:34:57 -- Running - OS: Windows, Chrome: 77.0.3865.75, Configurator: 10.6.0
2019-09-14 @ 15:34:58 -- Using cached release information for configurator releases.
2019-09-14 @ 15:35:11 -- Serial port successfully opened with ID: 1
2019-09-14 @ 15:35:11 -- MultiWii API version: 1.42.0
2019-09-14 @ 15:35:11 -- Flight controller info, identifier: BTFL, version: 4.1.0
2019-09-14 @ 15:35:11 -- Running firmware released on: Sep 5 2019 20:03:32
2019-09-14 @ 15:35:11 -- Board: EXF722DUAL, version: 0
2019-09-14 @ 15:35:11 -- Unique device ID: 0x5400403237510131313935
2019-09-14 @ 15:35:11 -- Craft name: iFlight XL6
2019-09-14 @ 15:35:11 -- Arming Disabled

This is were I click to Activate Booloader in setup

2019-09-14 @ 15:35:17 -- Unrecoverable failure of serial connection, disconnecting...
2019-09-14 @ 15:35:17 -- Failed to close serial port
2019-09-14 @ 15:35:20 -- Using cached builds information for jobs.
2019-09-14 @ 15:35:21 -- Using cached release information for firmware releases.

clicked button to flash
2019-09-14 @ 15:37:08 -- USB device successfully opened with ID: 1
2019-09-14 @ 15:37:08 -- USB device successfully closed
nothing happens, no flash

If I used bootloader button on FC
I did not get the serial connection error
but same result with open/close device instantly

MBoerlage, 2019-09-21:
I have the same issue - CLRacingF4, BFConfig 10.6 RC1 that I downloaed yesterday (318?). Have it in DFU mode (using ImpulseRC driver fixer). After I click on Flash, nothing happens. In the top left corner it says "usb connection succesfully closed). 

My workaround it use 10.5 to flash the firmware and then get back into 10.6 to set it up.

IvoFPV, 2019-09-22:
@mboerlage Use the release version RC1 of the configurator.

closed 2019-10-29 by stale[bot]