betaflight/betaflight-configurator #1660: PID freeze in build #310

cusglider, 2019-09-15:
As I moved around in the PID page I was not able to leave as shown in the video

cusglider, 2019-09-15:
https://www.dropbox.com/l/scl/AABBn6IWhmDx00I6bDpjijmbyCd2bQWfZwM

IvoFPV, 2019-09-15:
I've tried to replicate it but it doesn't happen when I do the exact same as you do. Can you do this always by following those steps? Or it only happens sometimes?

cusglider, 2019-09-15:
I clicked on the PID page once, looked at the rates subpage and noticed I got stuck when I wanted to move on. That's when I started the recording to show I got stuck. I actually only work with the configurator when I need to update the quad's firmware.

IvoFPV, 2019-09-15:
@cusglider Since you've had issues last time with firmware flashing and it fixed itself by reinstalling the configurator, could you try to install this one https://www.dropbox.com/s/6f68m033ytbv0bm/betaflight-configurator-installer_10.6.0_win32.exe?dl=0
This is the latest master release build.

cusglider, 2019-09-15:
I am currently using the chrome app, however will follow this link

McGiverGim, 2019-09-16:
I've been stuck sometimes at some tabs too, in a random way (I've not been able to replicate). I fixed some bug related with this in the past but it seems there are more.

IvoFPV, 2019-09-16:
@cusglider I just got it to lock up once. Still don't know what causes it. There is nothing showing up and it is completely random.

bizmar, 2019-09-19:
It happens to me only with the PID tab, about once a week. There's no repeatable set of actions... Haven't caught it yet with a debug version of the config.

closed 2019-10-26 by stale[bot]