betaflight/betaflight-configurator #1661: PID tuning sliders should not turn off and reset when adjusting Yaw P and I

evilmerlin, 2019-09-16:
Currently, making changes to Yaw P and I values will turn off and reset the sliders. Can this be changed such that it doesn't? The sliders don't affect Yaw values anyway. 
It's a bit tedious when tuning via the sliders and having it all go away because I made a small change in Yaw.

mikeller, 2019-09-16:
@IvoFPV: What's your thoughts on this?

IvoFPV, 2019-09-16:
@evilmerlin Sliders do affect yaw P and I values, the master slider does that. So it is not possible to have them not disable on manual input.
>It's a bit tedious when tuning via the sliders and having it all go away because I made a small change in Yaw.

What do you mean by this? Which values are you trying to change on yaw and it is not possible to do that with sliders only?

evilmerlin, 2019-09-16:
Actually I was only using the PD Balance, Gain and Stick Response Gain sliders.
The PD balance and gain sliders don't move the Yaw values, well at least the version I'm using doesn't. I know the Stick Response one changes FF values for all axis which is fine.

So I tuning via the sliders and my quad has this quirk where the default Yaw P is too high. I'll be tuning the Roll and Pitch via the balance and gain sliders, then make a change to Yaw P (which then turns off the sliders). Re-enabling sliders if I want to make changes moves my Yaw P back to default values. 
Just a small thing and was wondering if it can be done.

IvoFPV, 2019-09-16:
@evilmerlin The thing with sliders is that they should cover almost every build, but if you have something really specific like it seems it is in your case then you will have to manually tune that. Yaw P will be adjustable with PD gain slider soon, so that will probably fix your issue and give you an opportunity to tune that specific quad better.
Could you maybe post your log so we can see what is wrong with your yaw axis and see what can be done and what is necessary to fix that?

evilmerlin, 2019-09-16:
I see! I look forward to that then. I must say that the sliders are a brilliant brilliant implementation. Makes tuning so much easier.

Hmmm, I don't have that log anymore, deleted it after it was kinda solved by lowering Yaw P. 
I did ask for some help over on the Betaflight Black box Log  group on Facebook though.
https://www.facebook.com/groups/291745494678694/permalink/652520531934520/

PR #1662 (2019-09-17): Tuning sliders yaw P fix
Closes #1661 We have found out that yaw P should be included in PD gain slider. Good thing I checked right now because I forgot to PR this.
closed 2019-09-17 by mikeller