betaflight/betaflight-configurator #1667: Flashing Firmware Bricks CL Racing F7

azixaka, 2019-09-16:
Trying to flash any 4.1 RC3/Dev firmware via Configurator 10.60 RC1 bricks the CL Racing F7 at "flashing..." stage.

Boot mode/full erase/no boot sequence doesn't hangs at flashing stage as well. The erasing stage completes though.

Connecting via Configurator 10.51 and doing a boot mode restore works fine. Flashing any other version succeeds as well.

There seems to be a bug introduced in 10.60.

IvoFPV, 2019-09-17:
@azixaka 
>Flashing any other version succeeds as well.

Flashing other targets on 10.6.0 or other configurator versions?

What do you mean by it gets bricked? Which target did you select? CLRACINGF7 or CLRACINGF7 (Legacy)

IvoFPV, 2019-09-17:
@mikeller Looks to me like unified config is using F745 target but CLRACING F7 is F722. Probably why it bricks in this case.
```
# Betaflight / STM32F745 (S745) 4.0.0 Mar 10 2019 / 21:49:53 (d6138c41e) MSP API: 1.41
board_name CLRACINGF7
```
[image]