betaflight/betaflight-configurator #1668: macOS 10.15 'catalina' - “Betaflight Configurator.app” is damaged and can’t be opened. You should move it to the Bin.

gretel, 2019-09-17:
i tried opening the app holding `ctrl` key and select open which worked to override the failing integrity check for some other apps like the nightly build of `openscad` .. but i just can not open this build.
the `spracing` fork one does open fine.  the `chromeos` build works too, but fails to flash any board.
[image]

maybe the certificate used to compile this release is invalid?

IvoFPV, 2019-09-17:
#1624 This is the problem
>As a reminder, Mac software distributed outside the Mac App Store must be notarized by Apple in order to run on macOS Catalina. To make this transition easier ... , we’ve adjusted the notarization prerequisites until January 2020.

gretel, 2019-09-17:
actually i have a developer version of catalina and am allowed to run interim-signed apps in general. i doubt this is actually the only reason.

mikeller, 2019-09-18:
@gretel: Have you tried downloading the application again - the error message you are getting suggests file corruption.

gretel, 2019-09-18:
@mikeller yip, three times. it's a dmg which has checksums, though.

mikeller, 2019-09-18:
@gretel: Weird - I tested running the app inside the .dmg before I uploaded the release - will test again after downloading it.

gretel, 2019-09-18:
@mikeller can verify the image fine:

[image]

gretel, 2019-09-18:
what i find most irritating that the

hold `ctrl` - click application icon - select `open`

method does not allow to override.. like it does for other non-notarized apps.

mikeller, 2019-09-18:
I'll try downloading / installing it just to make sure. But I suspect it could be related to the fact that I've built this on an older version of MacOS.

gretel, 2019-09-18:
@mikeller i'm not so much an xcode expert but i suppose you need to build for `10.13` at least for the application to have a supported signature on `10.15` - is the build process documented? i could try..

gretel, 2019-09-18:
wait. wasn't that a `node` app wrapped somehow? and is xcode required at all?

mikeller, 2019-09-18:
Yes, it is an [NW.js](https://nwjs.io/) app. Xcode isn't required for the build itself, but the tool that is required to create a compressed .dmg comes only with xcode.

https://github.com/betaflight/betaflight-configurator#development has the build instructions.

gretel, 2019-09-18:
@mikeller thank you. it compiled like a charm. did image the result up - which works fine for me:

https://svaha.jitter.eu/betaflight-configurator_10.6.0-RC1_macOS_gretel.dmg

mikeller, 2019-09-18:
@gretel: Can you please test if this is working in Catalina:

https://www.dropbox.com/s/weo946gimcvjhkq/betaflight-configurator_10.6.0-RC1_macOS.dmg?dl=0

(Built in Mojave and provided by @ctzsnooze, thank you very much.)

gretel, 2019-09-18:
@mikeller does not, no observable difference. could you test the one i built on earlier osx?

[image]

etracer65, 2019-09-18:
@gretel Can you try this please?

https://drive.google.com/file/d/112oudwlcxhBkU02GiSLQ_4vH6UaCdQMJ/view?usp=sharing

It's just the app removed from the .dmg and zipped. Trying to determine if the problems are related to the disk image or with the app itself.

gretel, 2019-09-19:
the screenshot i posted shows that the disk image is fine. accordingly, the application you uploaded does not contain any valid signature and will not open.

gretel, 2019-09-19:
can somebody please try the build i did upload above? it just works for me.

mikeller, 2019-09-19:
@gretel: If you are not willing to cooperate with the developers you should not be opening issues, or expect to get help. Closing.

closed 2019-09-19 by mikeller

gretel, 2019-09-19:
@mikeller you must be kidding me. not cooperating, by building and uploading an application to test? i was just asking for somebody to try this. better check your attitude. this is not your project, but a community one.

gretel, 2019-09-19:
@mikeller you seem to lack respect. please check your attitude. looks like you do not know how to fix this issue and blame it one somebody else.

mikeller, 2019-09-19:
@gretel: So me telling you that you should not open issues here if you are not willing to cooperate with developers is 'lack of respect'? You really need to check your attitude. But please just stay away from here - people with your attitude are not a good match for open source projects.