betaflight/betaflight-configurator #1669: Github API Limit

richard-scott, 2019-09-17:
Looks like issue #61 is back.

[image]

Docteh, 2019-09-17:
Github API allows 60 unauthenticated requests per hour from a given IP address. The configurator does two or three and caches the results for an hour. Issue #61 occurred because there was zero caching.

Are you reporting that the caching stopped working?

richard-scott, 2019-09-24:
Yep, this error was from the first time I'd started the App that day after updating it from the Master branch.

Docteh, 2019-09-26:
2019-09-25 @ 19:48:48 -- Using cached release information for configurator releases.

Caching works on my end, but if it doesn't ever get a good response back from github, the configurator wont cache it. If its persisting past an hour it might be worth running down. I'm curious as to what other app would hit the same API

richard-scott, 2019-09-28:
I need a way to "authenticate" with Git so that I can use more API calls in BFC
I use VSCode and Git regularly and this is not working nicely with the BFC.

mikeller, 2019-09-28:
@richard-scott: Feel free to open a pull request to add optional authentication with GitHub to Betaflight configurator.

richard-scott, 2019-09-28:
It's odd tho, as all my Github usage _IS_ authenticated. As far as I'm aware, BFC is the only app that has so far reported this issue with the API.

If I can sort out a patch I will, but don't hold your breath :-)

mikeller, 2019-09-28:
@richard-scott: The thing is that the Betaflight configurator is not running inside chrome, so your cached session information for GitHub isn't available to it - you'd have to add code to log into GitHub from within the configurator.

richard-scott, 2019-10-15:
Maybe not. You could create an OAuth Token in the Developer Settings -> OAuth Apps section of the BFC Organisation and use that in the URL request for data:

> curl https://4ed460083d963434cf76:7757fd7086438bc2ea0bc3cd9a537e9ea790aeb8@api.github.com/repos/betaflight/unified-targets/contents/configs/default

That may not be the right way, but that URL worked for me until I cancelled the Token.

mikeller, 2019-10-15:
@richard-scott: Feel free to open a pull request to add a facility to allow users to add their own OAuth tokens to configurator - I don't know if this will see much use though, as combined with the caching we do the GitHub IP based limits seem to be fine for most users.

Docteh, 2019-10-15:
> You could create an OAuth Token in the Developer Settings -> OAuth Apps section of the BFC Organisation and use that in the URL request for data

OAuth gives full permissions of the user that generates the key, so hopefully that linked token is actually cancelled.

richard-scott, 2019-11-01:
@Docteh - Yes, that token has been deleted.

@mikeller - Maybe its worth someone looking at the caching logic because when my download from the API failed, I could not flash with BFC as the dropdown boxes were empty. I had previously used the app the day before, so cached data was/should have been available locally.

mikeller, 2019-11-01:
@richard-scott: We do have caching for pretty much everything we know to be affected by the API request limiting - so unless you can supply a reproduction of what limiting you were affected by there is not much we can do.
There also was an outage in AWS that affected some files that GitHub is hosting there - it is possible that the failure you got was caused by this.

closed 2019-12-09 by stale[bot]