betaflight/betaflight-configurator #1671: Video Transmitter tab not displaying pit mode status after a change

etracer65, 2019-09-18:
I found a problem in the Configurator Video Transmitter tab relating to pit mode. But it's not going to be a really simple fix. The issue is a timing problem that will cause the pit mode to not be indicated in the Configurator after enabling it and saving. The Configurator is not really at fault because it's displaying what the MSP says, but it's a timing issue. When the Configurator sends the `MSP_SET_VTX_CONFIG` message to enable pit mode the firmware correctly receives this and sends the command to the VTX to change into pit mode. The Configurator then immediately requests `MSP_VTX_CONFIG` to get the updated settings but the pit mode flag is based on the VTX reporting that it's in pit mode. Because communication with the VTX happens through (relatively) slow polling we haven't got the updated real status from the VTX yet so the MSP returns that pit mode is not enabled (even though the VTX is in the process of switching). If you leave the Video Transmitter tab and then return the pit mode will be properly displayed because enough time has elapsed that the firmware has communicated with the VTX and recognized the updated pit mode indication.

So the only way I can see to work around this is to have the Configurator periodically poll the `MSP_VTX_CONFIG` message and update it's state. Polling at a relatively slow rate would probably be ok - maybe 0.5 second or 1 second.

Doing a fixed delay after saving changes before sending the `MSP_VTX_CONFIG` message won't work because it's unknown how long it will take the VTX to actually make the change and the vtx control polling between the firmware and VTX is quite slow (could be 0 to 2 seconds depending on where we catch it in the cycle).

haslinghuis, 2021-06-05:
@etracer65 is this still an issue as a I'm sweeping through old issues getting resolved

nerdCopter, 2022-10-06:
I found this as well for EmuF.  My workaround (and admittedly bad practice) was to refresh after 3 seconds as i believe a pilot would not modify and save within such a short time.
https://github.com/emuflight/EmuConfigurator/blob/6f15c4dd6d48f8c60c2fe845b48ff47dacc423d0/src/js/tabs/vtx.js#L304

haslinghuis, 2022-10-06:
Lols. @nerdCopter Have made the same workaround.

PR #3034 (2022-10-11): Fix VTX status update after saving
Fixes: #1671