betaflight/betaflight-configurator #1673: Battery low indicator in the configurator

Etruscian, 2019-09-18:
**Is your feature request related to a problem? Please describe.**
When working on the bench, I usually have a battery connected. I have set up all the settings in betaflight so that it watches my battery voltage. Since the configurator does not show these warnings, but does show a green battery bar if a battery is connected, I've almost killed a lipo because I watched the green bar instead of the voltage.

**Describe the solution you'd like**
I would like the green battery bar to turn red when the connected battery is almost empty when attached to the configurator. Maybe also a warning popup or something like that

**Describe alternatives you've considered**
I've thought about battery voltage meters but since I don't have one, I'd like to have this feature in the configurator/betaflight. Since I also didn't have a receiver connected, I had no other way of giving me a warning.

PS: I didn't know if this was the right place for this kind of feature request. If it isn't, please tell me where I should put it and I'll happily close this request

McGiverGim, 2019-09-23:
@Etruscian in theory, it must turn red:

https://github.com/betaflight/betaflight-configurator/blob/2395c7a216ca0302554300c5af36d290ffec0391/src/js/serial_backend.js#L649-L680

It is done in a manual way (now we have better MSP commands to do this than the old ANALOG) but it must work.

The only problem I see is that it shows a red bar when it is in warning range, but if it is under the minimum there is no bar to show, because it ranges only from minimum to maximum..

Etruscian, 2019-09-23:
@McGiverGim the weird thing was that I did see a green bar that was almost full. Could this be that because the battery was empty before plugging in, it mistakenly was recognized as an over full 3s?

mikeller, 2019-09-23:
@Etruscian: Did you calibrate the voltage meter in the firmware? They can be considerably off.

McGiverGim, 2019-09-23:
@Etruscian the code divides the `voltagedetected / vbatmaxcellvoltage` to get the number of cells. If it detects the wrong number of cells yes, it can go wrong.

I will push a change to blink when the voltage is under minimum, but if the cell count is not correct, I can't do nothing.

closed 2019-10-14 by mikeller