betaflight/betaflight-configurator #1674: Using DFU Boot configurator button crashes Aikon F4

SupaflyFPV, 2019-09-19:
I have a problem where when trying to use the configurator boot/DFU button with the Aikon F4 flight controller - the configurator tries to reboot but  FC appears to crash...mouse behaviour goes weird and no more action possible on the main configurator screen...I have to unplug - replug...it won't go into DFU mode...this is on Mac with 10.5.1 and newest Jenkins build...

My physical boot button is not easily accessible hence the effort...

mikeller, 2019-09-19:
@SupaflyFPV: Do you have any peripherals connected on any of the UARTs (RX, GPS, VTX, ...) that are powered while you are trying to get into DFU mode? If so, disconnect them from the board, or disconnect the power to them, as they will probably interfere with the MCU binding to USB in DFU mode.

SupaflyFPV, 2019-09-19:
no just the FC connected...

mikeller, 2019-09-19:
@SupaflyFPV: Can you try to download and install configurator 10.6.0-RC1 (https://github.com/betaflight/betaflight-configurator/releases/tag/10.6.0-RC1) and check if this version works?

SupaflyFPV, 2019-09-19:
@mikeller unfortunately it does the same thing...the behaviour once boot mode is attempted is the same co-incidentally as I had here

https://github.com/betaflight/betaflight/issues/8840

Not sure if that is a generic 'crash' behaviour but thankfully I don't have to reflash - I just unplug the FC and reconnect it...

mikeller, 2019-09-19:
@SupaflyFPV: 

> the behaviour once boot mode is attempted is the same co-incidentally as I had here betaflight/betaflight#8840

Not sure I understand this - betaflight/betaflight#8840 is a firmware issue, whereas in this ticket you describe a configurator malfunction.
If this is a configurator issue, is it possible to get a video of the way it misbehaves?

SupaflyFPV, 2019-09-19:
I just meant that in the above issue, the behaviour which boots out of the configurator (the other issue boots out after saving in CLI, and this one when pressing the software DFU button) leading to a situation where you cannot log back into the configurator, and producing a laggy, stuttering mouse behaviour...was the same...I mentioned it in case it could be connected

mikeller, 2019-09-19:
Well the laggy, stuttering mouse behaviour is definitely a configurator problem - I've seen similar things happen when something was messed up with the WebGL driver, but usually a complete uninstall / reinstall of configurator fixes this.

closed 2019-10-26 by stale[bot]