betaflight/betaflight-configurator #1675: can't open on mac

trojanGoat, 2019-09-19:
I just tried installing 10.6 configurator on my mac and it seems like the os is detecting it as a damaged file

[https://i.imgur.com/0kwxHvU.png](url)

mikeller, 2019-09-19:
@trojanGoat: What version of MacOS are you using?

trojanGoat, 2019-09-19:
I'm using Mojave 10.14.6

mikeller, 2019-09-19:
@trojanGoat: Can you please check if this works for you: https://www.dropbox.com/s/weo946gimcvjhkq/betaflight-configurator_10.6.0-RC1_macOS.dmg?dl=0

trojanGoat, 2019-09-19:
No it doesn't work.  Same issue.

IvoFPV, 2019-09-19:
Isn't this related to it being unsigned? Even though it says that it is damaged, macOS throws this exact error when you try to install unsigned apps and don't enable the option to `Allow apps downloaded from: Anywhere` which can't be done through GUI in newer macOS versions. Only way to put it to `Anywhere`, disabling "Gatekeeper", is through terminal.

trojanGoat, 2019-09-19:
When I run 10.5 configurator I have the option to allow the app to run in the security settings.  The OS doesn't say that it's damaged.

IvoFPV, 2019-09-19:
What does clicking on the question mark button show? It should show the reason it won't open.

trojanGoat, 2019-09-19:
@IvoFPV  This is what it says:

> The app has been modified or damaged
> The app has been modified, and its code does not match the original signed code. The app may be broken or corrupted, or it may have been tampered with. If you think that an app has been damaged or tampered with, you should delete it by dragging it to the Trash.
> 
> Some apps and tools, such as AppleScript or JavaScript applications and some legacy tools, modify themselves after signing. These types of apps cannot be opened unless you override the security settings on your Mac. See Open an app by overriding security settings.

mikeller, 2019-09-19:
@trojanGoat:

> Some apps and tools, such as AppleScript or JavaScript applications and some legacy tools, modify themselves after signing. These types of apps cannot be opened unless you override the security settings on your Mac. See Open an app by overriding security settings.

That's pretty much what this is I suspect. The NW.js app runner that we are using must be signed, and obviously we modify the app by adding our code to be run. Looks like the way to fix this is to override the security settings for the Betaflight configurator on your machine.

emski0, 2019-09-19:
Hey i am getting also file is damaged and can't be opened (macOS Mojave 10.14.6 ). However if i build myself the app from source, it is working.
Try this , i build it from current source.
https://www.dropbox.com/s/7slm5wvdwb3qrfn/betaflight-configurator.app.zip?dl=0

mikeller, 2019-09-19:
@emski0: See https://github.com/betaflight/betaflight-configurator/issues/1675#issuecomment-533126426.

trojanGoat, 2019-09-19:
How come it works for 10.5 configurator?

mikeller, 2019-09-19:
@trojanGoat:

> How come it works for 10.5 configurator?

Not sure really. This is something that apple does to detect 'corrupted' apps, and  you'd have to ask them what rules they use for this. The 10.6.0-RC1 version in dropbox.com/s/7slm5wvdwb3qrfn/betaflight-configurator.app.zip?dl=0 was built on Mojave - so if this does not work but a locally built version does then that suggests that the detection logic is exempting locally built apps from being flagged. Unfortunately this is no help for us.

vinc13008, 2019-09-19:
just disable gatekeeper with the Terminal : 
sudo spctl --master-disable

Docteh, 2019-09-19:
I think one unanswered question from this, is

Has anyone recently tried installing the 10.5.1 configurator on a macbook that has NEVER touched the configurator?

IvoFPV, 2019-09-19:
Also one thing to add, when we were testing the sliders a few weeks ago and making builds for Mac they worked just fine. CTZ and Bizmar built them. I don't see what is different about the RC1 build...
Here is one of them, a bit outdated but maybe just to test out if this one is any different.
@trojanGoat @emski0 Could you try it out?
https://www.dropbox.com/s/pu25aau8v8w71fg/betaflight-configurator_10.6.0_macOS.dmg?dl=0

Propwosh, 2019-09-20:
> just disable gatekeeper with the Terminal :
> sudo spctl --master-disable

Had the same issue..but This worked for me! Thanks!

trojanGoat, 2019-09-21:
> > just disable gatekeeper with the Terminal :
> > sudo spctl --master-disable
> 
> Had the same issue..but This worked for me! Thanks!

This works!! Thanks

ddckl76, 2019-09-21:
> sudo spctl --master-disable

THIS WORKS! Had the same issue with the unsigned certs. No previous versions of configurator had this issue, mainly because there was an option to 'right click' and 'open' instead of 'double click(ing)' the icon to open the app.

mikeller, 2019-09-21:
@ddckl76:

> No previous versions of configurator had this issue, mainly because there was an option to 'right click' and 'open' instead of 'double click(ing)' the icon to open the app.

I am not a Mac user, so I do not know exactly what the difference between right click / 'Open' and double clicking to open is on a Mac. But this is not something that we changed - it sounds more like this was changed in the NW.js toolkit that we are using to build the configurator. As such, this is not something that we can change.

Docteh, 2019-09-21:
I thought we are using the same version of nwjs as the last release. (10.5.1) On my MacBook I need to right click and open any app downloaded from the internet, just double clicking the first time I get a complaint from macOS. After I allow it once then everything is fine. I wonder if the beta of Catalina has trouble keeping track of where unsigned apps come from. I downloaded the rc1 configurator from GitHub and it worked fine but maybe I'm one version behind.

mikeller, 2019-09-21:
@Docteh: You are right, we did not update the NW.js version in the end - so that can't be the reason.

schmou187, 2019-09-22:
It worked aswell for me (with a older version of OS 10.13) 

> sudo spctl --master-disable

thanks alot

mrRobot62, 2019-09-23:
I'm using MOJAVE 10.14.6 on a MacBook Pro with 16GB Ram - installation process worked as in the past - so no problems on my system. I didn't do anything "on top" to launch or install the software.

ppporch, 2019-10-01:
I had the same issue. RC1 installed and asked me if I wanted to open an unsigned app. RC2 gave me the error message of the file being damaged and wanted me to move it to the trash. The sudo spctl --master-disable allowed me to install it and open it. I then went back in and turned the security feature back on with sudo spctl --master-enable

mikeller, 2019-10-01:
@ppporch: Thanks for the feedback. So to clarify, after you'd installed Betaflight configurator with `sudo spctl --master-disable`, you were still able to use configurator even after you ran `sudo spctl --master-enable`?

trojanGoat, 2019-10-05:
@mikeller Yes the configurator will continue to work after running 
> sudo spctl --master-enable

mikeller, 2019-10-07:
Added note to the release notes and to the README (#1723). Thanks all for helping to develop this workaround.

Letzrock, 2019-10-11:
Using: sudo spctl --master-disable  didn't work for me using Catalina

10.6 config
Catalina 10.5

mikeller, 2019-10-12:
@Letzrock: Can you try if this makes the installation work for you:

```
sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app
```

(and if not, is there an error message when you try to run the command, and what is it?)

Letzrock, 2019-10-12:
Yes that does work. AJ Christian gave me that info yesterday on FB

@Letzrock: Can you try if this makes the installation work for you:

sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app
(and if not, is there an error message when you try to run the command, and what is it?)

fichek, 2019-10-16:
> @Letzrock: Can you try if this makes the installation work for you:
> 
> ```
> sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app
> ```
> 
> (and if not, is there an error message when you try to run the command, and what is it?)

Yes, that's a proper way to work around this for now.

closed 2019-11-23 by stale[bot]