betaflight/betaflight-configurator #1676: PD Balance only changes P?

Knifa, 2019-09-19:
As the title says.

On the Windows RC1 build

The name and description seem to imply both should change, but only P changes.

IvoFPV, 2019-09-19:
This is the intended behavior. PD balance slider changes the ratio between P and D. You could change either P or D to affect that ratio, we chose to only change P. Changing both would change D and we don't want that because of how D is affected by and affects noise levels. That's why there is a P and D gain slider which moves both up or down.
Description fits the slider in my opinion really good. If you think otherwise, could you point it out?

Mateyhv, 2019-09-24:
@IvoFPV the description is not bad but can be misleading as happened to Knifa. Maybe "P to D ballance" will make easier to understand why is P only moving. Just a suggestion though

Knifa, 2019-09-24:
After playing around I did figure out what was meant. :smile:
And I get what you mean --- it makes sense to just change one rather than changing one up, and the other down.

I like the suggestion of naming it "P to D Balance" instead, though we're probably splitting hairs at this point.

IvoFPV, 2019-09-24:
We can do that. If it makes it clearer to what it does at least to someone it is a good change 👍

closed 2019-10-31 by stale[bot]