betaflight/betaflight-configurator #1679: Pasting diff doesn't update settings in CLI Configurator build #318 chrome OS

Gozzz, 2019-09-20:
Configurator build #318 chrome OS on a mac.

When attempting to paste a diff in CLI it seems to quit after "defaults nosave" line. Pasting partial diff starting at #resources completes fine.

See attached screenshot.
[image]

IvoFPV, 2019-09-20:
@Gozzz If I understand correctly, `defaults nosave` causes it to reset before other settings therefore they don't get applied. I tired to replicate the issue but it doesn't happen. I tried on RC1 configurator build and F405 target.

Gozzz, 2019-09-20:
Correct, it hangs at the point in my screenshot and doesn't process anything else. Could it be target related? This is a TalonF7v2. Diff is attached.

If I cut everything above #resources it will process it fine.

[4-1 default for A1.txt](https://github.com/betaflight/betaflight-configurator/files/3637536/4-1.default.for.A1.txt)

IvoFPV, 2019-09-20:
Could you remove the `defaults nosave` line and try again?

Gozzz, 2019-09-20:
@IvoFPV Removing that line it works as intended.

IvoFPV, 2019-09-20:
Not sure, I even tried your exact diff and everything goes on after `defaults nosave`, no hang up there. So it doesn't lock up or anything, you can continue typing other commands manually after that happens?
Could you do `defaults show` and paste it here?

Gozzz, 2019-09-20:
Odd, I will try a different mac with the same diff and see what happens.

defaults show-
`# defaults show
# Betaflight / STM32F7X2 (S7X2) 4.0.5 Aug  1 2019 / 12:29:33 (839661b20) MSP API: 1.41
board_name TALONF7V2
manufacturer_id HENA

# resources
resource BEEPER 1 B04
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 C08
resource MOTOR 5 A01
resource MOTOR 6 B09
resource MOTOR 7 C09
resource MOTOR 8 B01
resource LED_STRIP 1 B01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource LED 1 B00
resource RX_BIND 1 B02
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 B03
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource PINIO 1 A14
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

# timer
timer B03 AF1           # pin B03: TIM2 CH2 (AF1)
timer B06 AF2           # pin B06: TIM4 CH1 (AF2)
timer B07 AF2           # pin B07: TIM4 CH2 (AF2)
timer B08 AF2           # pin B08: TIM4 CH3 (AF2)
timer B09 AF2           # pin B09: TIM4 CH4 (AF2)
timer A01 AF2           # pin A01: TIM5 CH2 (AF2)
timer C08 AF3           # pin C08: TIM8 CH3 (AF3)
timer C09 AF3           # pin C09: TIM8 CH4 (AF3)
timer B01 AF2           # pin B01: TIM3 CH4 (AF2)

# dma
dma ADC 1 1             # ADC 1: DMA2 Stream 4 Channel 0
dma pin B03 0           # pin B03: DMA1 Stream 6 Channel 3
dma pin B06 0           # pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0           # pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0           # pin B08: DMA1 Stream 7 Channel 2
dma pin A01 0           # pin A01: DMA1 Stream 4 Channel 6
dma pin C08 0           # pin C08: DMA2 Stream 2 Channel 0
dma pin C09 0           # pin C09: DMA2 Stream 7 Channel 7
dma pin B01 0           # pin B01: DMA1 Stream 2 Channel 5

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature OSD

# serial
serial 0 2048 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# aux
aux 0 39 0 900 2100 0 0

# master
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_device = SPIFLASH
set dshot_burst = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 160
set ibata_scale = 150
set beeper_inversion = ON
set beeper_od = OFF
set pid_process_denom = 1
set osd_vbat_pos = 2447
set osd_rssi_pos = 2074
set osd_tim_2_pos = 2456
set osd_vtx_channel_pos = 2049
set osd_current_pos = 2439
set osd_mah_drawn_pos = 2433
set osd_craft_name_pos = 2058
set osd_warnings_pos = 14378
set max7456_spi_bus = 3
set dashboard_i2c_bus = 2
set flash_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW0
set pinio_box = 39,255,255,255

IvoFPV, 2019-09-20:
Yeah, that looks okay. Maybe try the official RC1 build, because there were some issues installation related with flashing when using Jenkins ChromeOS builds.

Gozzz, 2019-09-20:
@IvoFPV I just tried it on a different laptop, Also mac OS with configurator build 318 same thing. 

Then I tried the RC1 dmg installer and it also does the same.

So I can continue to type into cli after it hangs but it doesn't return anything. For example typing "get vtx" will not return anything.

IvoFPV, 2019-09-20:
@Gozzz Thanks for testing it. If you have time and have a different FC or even better one with F4, try the same thing to see if it's target specific. Not sure what else could be causing it.

Gozzz, 2019-09-20:
@IvoFPV so it works fine on the clracingf4 unified & legacy targets. 

I also tested it on a clracingf7 unified target and it will hang same as the talonf7v2. 

Tested on the same clracingf7 using legacy target and it imports fine.

Unfortunately those are the only f7 boards I have on hand to test with so I can't narrow it down any further.

IvoFPV, 2019-09-20:
Thanks for confirming that it works on F4 target. @mikeller Any thoughts on what could be causing this?