betaflight/betaflight-configurator #1680: BFC 10.6 RC1: Fonts look smashed... :-/

RipperDrone, 2019-09-21:
[image]

Docteh, 2019-09-21:
Font rendering issues?

* What resolution is that monitor set to?
* Whats the DPI setting, mine is 100%
* Version of windows, I'm on 1903
* Is this the RC1 build we have posted on github, or did you build this yourself?

RipperDrone, 2019-09-21:
Not really font rendering, screen somehow flickers when scrolling BFC window scroll bars, some areas seem a little clearer, but most content is dithered. Doesn'T happen at all with Nightlies installed via Chrome...?

DPI 100%, RC1 BFC from GitHub, Win 10 x64 1903

Docteh, 2019-09-21:
When I looked at the screenshot on github it just looked like the text rendering was mucked up, but then I looked closer. 
[image]

The [NWJS](https://nwjs.io/) that runs the configurator stand alone, is really just a packaged copy of the chrome browser. The [build](https://nwjs.io/blog/v0.36.4/) we're using is dated 2019-03-04. I tried looking up blurry apps, and most results are from 2015. 

Personally I'd try running the latest [INAV configurator](https://github.com/iNavFlight/inav-configurator/releases), and see if it exhibits the same issue. Their latest 2.2.1 release is running NWJS 0.36.4 as well, so its the exact same program (NWJS, not the actual app), just packaged differently. Grab the win32 since we're win32.

Otherwise you're immediately left with trying things such as wiping the `betaflight-configurator` directory in `%LOCALAPPDATA%`, fully uninstalling then reinstalling the configurator, or looking at options in windows 10. I noticed that there is an option for windows to try and fix blurry apps. I tried turning it off, but the configurator still looks fine for me.

RipperDrone, 2019-09-22:
@Docteh HAve done complete uninstall / reinstall of BFC, plus reset my GF1060Ti graphics card to 'quality' preference. PRoblems gone since. Loaded iNAV, fonts and windows stretching / scaling behavior look the same (slightly dithering the fonts while stretching/scaling but never completely wash them out...). Seems to be graphics drivers or interference of Chrome version vs. standalone version installs / libraries used... closing this issue now...

closed 2019-09-22 by RipperDrone

oermenz, 2019-10-15:
I have the exact same problem as above, and had the same problem with "Discord". What fixed it there for me was to turn off hardware acceleration, but that isn't within "Betaflight" options. I'm going to check if changing some videocard setting will change this issue. Changing DPI settings did nothing for me. Only had two programs having this issue, betaflight and discord.

RipperDrone, 2019-10-15:
@oermenz good to have someone replicate the issue. Graphics card is Nvidia, too?

mikeller, 2019-10-15:
@RipperDrone, @oermenz: Try the `--disable-gpu` option when starting configurator: http://docs.nwjs.io/en/latest/References/Command%20Line%20Options/#-disable-gpu

oermenz, 2019-10-15:
@mikeller This seems to have done the trick! Ty!

oermenz, 2019-10-15:
After some more searching it seems to be an issue with Electron.js that it also used by Discord. The problem seems to be with Nvidia FXAA setting. Another option is to add the program in the "manage 3d setting" tab within the Nvidia control panel, and set FXAA to off for the specified program.

RipperDrone, 2019-10-15:
@mikeller @oermenz 1k kudos! Did the trick! Wow, that was a hard one :-). 

[image]

RipperDrone, 2019-10-15:
So it seems we need the command line gpu and/or 'no Anti-aliasing' preset as a default :-)

oermenz, 2019-10-15:
> So it seems we need the command line -no_gpu and/or 'no Anti-aliasing' preset as a default :-)

That seems a bit radical since I don't think many users have ran into this problem. The solution is simple for people having to bypass it.

mikeller, 2019-10-16:
@RipperDrone, @oermenz: Does anybody want to collect an easy point for Hacktoberfest (https://hacktoberfest.digitalocean.com/) with a pull request to add a note to https://github.com/betaflight/betaflight-configurator#webgl telling users to try `--disable-gpu` when they run into display problems?

RipperDrone, 2019-10-16:
@mikeller sounds simple - I registered on Hacktoberfest, then tried to open a PR here, realized that it needs me to install the whole coding environment and branching out my own branch first. Man, this is just so much overhead for doing a simple text edit PR :-/

McGiverGim, 2019-10-16:
You can edit the text directly in Github, and create the pr directly there. You need to have a fork of the repo, press edit in the file you want to modify here in Betaflight, and you will see a create a branch button.

McGiverGim, 2019-10-16:
You have done a PR to your own repository 😋

Select your branch, and do a new PR, and be sure to select the Betaflight repo as target for the PR.

RipperDrone, 2019-10-16:
@McGiverGim better now?

oermenz, 2019-10-20:
@RipperDrone When I searched for the problem I came across another issue you've opened about the same subject which is still open. I think you can close that one, or remove completely. No need for issue cluttering! ;)