betaflight/betaflight-configurator #1681: BFC 10.6 RC1: Switch indicators misplaced (Win10)

RipperDrone, 2019-09-21:
[image]

IvoFPV, 2019-09-21:
@RipperDrone When opening an issue, please provide more information about what exactly is the problem and how can we replicate it. Also you should make sure it is not an issue on your side. I would suggest you completely reinstall the configurator with the RC1 release version. 
This does not happen when I test it.

RipperDrone, 2019-09-21:
I didn't reinstall but tried many times to replicate, but now any time it showed ok. Seems there is a certain sequence needed to hit that issue. I did reinstall the complete BFC package (Win32 version RC1), but at the same time the Chrome BFC was still installed. I then de-installed the Chrome version. Now all looks good...?

closed 2019-09-21 by RipperDrone

Docteh, 2019-09-22:
Should #1680 be closed as well? If we get another report of bad graphical rendering in the future we can always try and run it down then