betaflight/betaflight-configurator #1684: Flashing FC : can't flash (sometimes) downloaded hex file into FC via "Button local hex..."

mrRobot62, 2019-09-23:
Configuration:
Macbook Pro, Mojave 10.14.6
BF 10.6.0RC1.
FC: BrainFPV Radix

Current BF version on FC: BF 3.5.7

Steps
1. Download BF 4.0.5 hex file from BrainFPV website with customized FW and save on local dis
2. Set FC into DFU-Mode
3. Use Button "Load FW local" and choose downloaded file (1)
4.1. a black bar on the left side appears with "Loaded Local Firmware: (454178 bytes)"
4.2 Button "Flash Firmware" is active
4.3 I do not use any filtering from above switches or comboboxes (like Board, FW version, ...)
5 click button "Flash"...

I tried this several times - without any success. After restarting BF - it works (I tried this several times, too)

**possible problem:**
if user change or click something on switches / comboboxes, it is not possible to flash a local file. If BF is restarted and I click on Load FW local and click directly on Flash FW - no problems flashing is possible

Docteh, 2019-09-23:
What sort of without any success? Is there an error message that shows? does the flashing progress bar stall in the middle?

mikeller, 2019-09-23:
@mrRobot62: Do you have any external devices connected to your flight controller's UARTs that are powered while you are trying to flash the firmware? If so, disconnect them or make sure they are not powered.

mrRobot62, 2019-09-24:
Hi,
"..without success":
If I pressed the Flash-Button, nothing happens, no Progress bar, no error message nothing. It seems, that nothing was done "behind" the button

"...external devices":
It was a "blank" FC without anything installed (out of the box).

As I wrote:
If I don't Change anything at the top Switches or comboboxes in BF 4.1 and I use Load FW local and press than the button "Flash..." - no Problems the flashing procedure works well

But if I do something at the top Switches - like Erase ... (I know it make no sense to do this) and press than "Load FW local" and than press button "Flash" nothing happens, no flashing process is done, no update on Progress bar.

Hope this is more understandable for you

IvoFPV, 2019-09-24:
@mrRobot62 
I've just tested flashing local firmware, without changing switches like you say and it works just fine. I am on RC1 as well, running on Windows. Are you running the Chrome app?
> After restarting BF - it works (I tried this several times, too)

What do you do before you go to flash? Do you connect to the FC, make some changes, then go to flash and only then it doesn't work?

McGiverGim, 2019-09-24:
I think this has happened some times to me, but I don't know why. My solution was to disconnect and connect the FC again, after that the flash button worked for me. I have not tested with latest version so I'm not sure if the problem remains.

mrRobot62, 2019-09-24:
> @mrRobot62
> I've just tested flashing local firmware, without changing switches like you say and it works just fine. I am on RC1 as well, running on Windows. Are you running the Chrome app?
>
>
> What do [...]

As I wrote : If I do not change anything  - it works well under my "stand alone" application (not CHRROME app).

I know it's nonsense to change anything via comboboxes (like FC-HW, BF FW version,..)  if I want to flash via local hex - but if a user do this - he/she is not able to flash because it didn't work - than - 

I assume there is a bad validation algorithm implemented (or there should a validation algorithm implanted) to avoid this behavior.

Means: if any changes are made via switches/comboboxes, Flash-Button is deactivated.

Flash-Button should only be activated if user click on Load FW (online) or Load FW local AND if button is activated and than user change something on switches the validator should disable the flash-button again.

IvoFPV, 2019-09-24:
@mrRobot62 Yeah, I understand now what you are saying. However, I was not able to reproduce it following your steps. I just click load local firmware and then flash firmware, I don't touch anything else and it works. Hopefully someone else can reproduce it and find what's wrong.

mrRobot62, 2019-09-24:
From my point of view as an ex-devleoper it should be more or less easy to solve this issue. But honestly it's an minor priority ;-)

IvoFPV, 2019-09-24:
@mrRobot62 Reading through the whole thing I think everyone else and I have totally misunderstood you. You are not having a problem with actual flashing and are concerned about the button 'Flash Firmware' and when it is enabled/disabled.
>Flash-Button should only be activated if user click on Load FW (online) or Load FW local AND if button is activated and than user change something on switches the validator should disable the flash-button again.

This is exactly how it works. Flash button gets enabled only when you: 
1. Load Firmware [Local] and that is successful.
2. Load Firmware [Online] and that is successful.

It gets disabled when you:
1. Click Load Firmware [Local] to load another local firmware and you don't select anything or that is unsuccessful.
2. Change Board(Target) or Firmware version you want to flash, but only when Local Firmware NOT loaded already.

This is how it should work and makes sense to me.

mrRobot62, 2019-09-25:
@IvoFPV: u're right, but it didn't. The flash button will not be gray again, if you change anything on the switches or comboboxes.

BTW: I assume the misunderstanding based on my bad english - I'm not a native english speak. Sorry for any confusions.

I can reproduce the scenario several times:

1) start FC into DFU mode - ok - 
2) Flash button should be gray, Load FW local (yellow) - ok - 
3) load local FW -> flash button yellow - flashing possible ! - ok -

Do the same but...
1) start into DFU
2) load local fw
3) change something with combo boxes
4) flash changes back to gray
5) load local FW again
6) flash button change to yellow
7) click flash button -> it will not work now

That's on my system

As I wrote - everybody is right - it make not really sense to change something like (choose release or release candidate) and click than "load local fw". But if - than, there is a little bug

IvoFPV, 2019-09-25:
@mrRobot62 
> The flash button will not be gray again, if you change anything on the switches or comboboxes.

It won't be gray if you have local firmware loaded as I've said.

@Docteh @mikeller While I was looking into this I got it to apply a unified config from the selected board when I flashed Local firmware. Is this intended? I know you can load the config locally and flash it with local firmware but wasn't sure if you can mix local firmware and online config. Maybe I am wrong.

mikeller, 2019-09-25:
@IvoFPV: Yes, that is intended. It is not well documented yet, and I am not sure there isn't a more intuitive way to do this, so I am kind of waiting for better idea.
But the use cases we are trying to cover is that a tinkerer wants to share firmware with a new feature with friends, or a Betaflight maintainer wants to enable users to test a fix. So they should be able to share Unified Target .hex files, and users should then be able to flash them with the config for their target in an easy way.

closed 2019-11-01 by stale[bot]