betaflight/betaflight-configurator #1685: Mobile support?

Niek, 2019-09-24:
Has there been any effort to port the configurator to a mobile (Android and possibly iOS) app? It seems that either Cordova or Flutter would be the best approach. It'll basically be a webview with a thin wrapper around the chrome.* calls. As far as I can see the following Chrome Apps APIs need to be ported:
```javascript
chrome.runtime.lastError
chrome.storage.local.get/set/remove
chrome.app.window.create/current
chrome.fileSystem.*
chrome.usb.*
chrome.serial.*
chrome.sockets.tcp.*
```
For Cordova, the following plugins already exist, although most of them are not actively maintained anymore:
https://github.com/MobileChromeApps/cordova-plugin-chrome-apps-runtime
https://github.com/MobileChromeApps/cordova-plugin-chrome-apps-storage
https://github.com/MobileChromeApps/cordova-plugin-chrome-apps-fileSystem
https://github.com/MobileChromeApps/cordova-plugin-chrome-apps-usb
https://github.com/r1000ru/cordova-plugin-chrome-apps-serial
https://github.com/KoenLav/cordova-plugin-chrome-apps-sockets-tcp

For Flutter or another framework, it would require creating the Chrome Apps API wrapper from scratch.

McGiverGim, 2019-09-24:
No, it has been talked several times but nobody has tried to do it. I don't have experience with it so I can't give an opinion.

It will be a great addition, but we need some requisites:
- The maintenance must be very low. We don't have too much time for it.
- It must use the sources of the standard Configurator, without the need of duplicate and modify each of them.

Do you have experience with this? Do you think is possible?

Anisotropic, 2019-09-24:
@niek Have you tried "Speedy Bee" application by Runcam? It does work via cable as well.

Mateyhv, 2019-09-24:
Yep, Speedy Bee works with Betaflight, Inav and Kiss and as a bonus you can change motors spinning direction, however it won't have the latest changes in Betaflight configurator

Niek, 2019-09-24:
I am aware of Speedy Bee, but that's closed source (CMIIW) and doesn't have the full/latest Betaflight Configurator functionality.

I'll try to build something based in Flutter (since the Cordova libs are deprecated) using the existing source and see how far it'll come.

McGiverGim, 2019-09-24:
Waiting your news about it 🤗

Docteh, 2019-09-25:
flutter looks like it uses a slightly different language, but this might still be worth looking at if shipping standalone apps is possible. is "flutter for desktop" a thing?

Niek, 2019-09-25:
Yes, flutter desktop was released recently. It's still in heavy development though and webviews are not yet supported. Ref: https://github.com/flutter/flutter/issues/31712 https://github.com/flutter/flutter/issues/31713 https://github.com/flutter/flutter/issues/31714

McGiverGim, 2019-10-29:
@Niek have you worked on this? Only to know, if not the stale bot will close this if we don't take some action ;-)

Niek, 2019-10-29:
I didn't have too much time to work on this, but there is some progress:

[image]

hsuq510, 2019-11-02:
@Niek wrote you on slack also, are you using https://github.com/xseignard/cordovarduino.git? I cant connect with the device with VIP/PID aligned also.

MaximeAmadasi, 2020-02-09:
I tried to port the configurator to android with Cordova and it works well. The main issue I found actually is the firmware flasher : the configurator detects dfu but is not able to flash the fc.
I created gulp tasks to build the cordova folder with the ressources of the classic configurator. Only the serial.js file is different. If you are interrested [https://github.com/MaximeAmadasi/betaflight-configurator/tree/cordova](url)

Asizon, 2020-02-09:
@MaximeAmadasi woow very good neews, how I can try it?

MaximeAmadasi, 2020-02-09:
You can try this apk https://github.com/MaximeAmadasi/betaflight-configurator/releases/tag/10.6.0
Actually yarn gulp cordova will build the cordova folder. But you need to install java jdk, android sdk and build manually with cordova command line. I try to find an easy way to build the app with gulp.
Moreover, I found an other bug, sometimes, after a "Save and reboot", you need to manually disconnect and reconnect the fc

Niek, 2020-02-09:
Good job @MaximeAmadasi! The Cordova route is definitely quicker to implement (although not as cross-platform, I don't think there is enough iOS support in the plugins to make it work on Apple devices).

BTW I think the "plugins" dir should not be committed to the repo, as those should be installed by NPM (to keep the changes as minimal as possible).

Niek, 2020-06-14:
I see release 10.7.0 has Android support (preview), is this a NWJS build for Android? I can't find any reference in this repo or NWJS.

McGiverGim, 2020-06-14:
It will be merged and released AFTER 10.7.0.

mikeller, 2020-06-14:
@Niek: It is a preview version built from https://github.com/betaflight/betaflight-configurator/tree/10.7.0-preview.

Thunfisch2, 2020-07-02:
Would be great to see a iOS version, too! Or are there any technical restrictions due to iOS?

WalcoFPV, 2020-07-02:
@Thunfisch2 It should be possible to make a ios version. But first we need to implement bluetooth connection in the configurator

joevo2, 2020-07-02:
> @Thunfisch2 It should be possible to make a ios version. But first we need to implement bluetooth connection in the configurator

I saw that configurator seems to support Bluetooth already right. It’s more of a Cordova thing?

joevo2, 2020-08-28:
> @molor No. There is no date. I began to work on it but I don't know when it will be finished

Glad to hear this! What kinda bluetooth module are you using on your FC to communicate with the app?

PR #2577 (2021-09-09): Added build for Android Play store bundle.
Fixes #1685.

Based on #2574, will need a rebase after this has been merged.

Included:
- [x] added support for debug builds for android, producing an APK file for local installation;
- [x] changed the release build for android to produce an AAB file for Play store upload;
- [x] changed the release build for android to use a private keystore;
- [x] changed the release build for android to ask for the version name to be used, to work around the limitations on versioning applications in the Google Play store.
closed 2021-09-09 by mikeller