betaflight/betaflight-configurator #1688: ESC/Motor protocol popup

0crap, 2019-09-24:
Is DSHOT 900 and/or 1200 still in use?
It's not in my pull-down menu.
Using BF configurator 10.6.0-RC1.

[image]

Docteh, 2019-09-26:
Wow, I'd never even heard of DSHOT900, I saw that DSHOT1200 got removed on the firmware 

https://github.com/betaflight/betaflight/pull/8806

0crap, 2019-09-26:
Me either, I think that's always been a typo and should read Dshot600.
The complete bold printed part has been removed in the fix.
Seems like the proper decision.

mikeller, 2019-09-26:
If I remember correctly there was Dshot900 for a short time, before Dshot1200 became a thing.

PR #1689 (2019-09-26): Fix esc protocol tooltip for removed dshot1200
Fixes #1688
Dshot1200 is not available for 4.1 therefore it is removed from the tooltip.
closed 2019-09-26 by mikeller