betaflight/betaflight-configurator #1691: Error while trying update to version 10.5.1 - Windows 10

gmumdzhiev, 2019-09-26:
[image]

I have tried to update my previous version , but when i download the .exe from the page it doesn't want to launch it . It throws me this error . Any thoughts ? What should i do ?

McGiverGim, 2019-09-26:
Maybe is some antivirus or the new anti ransomware protection of Windows 10. Do you have it enabled?

gmumdzhiev, 2019-09-26:
@McGiverGim I don't use an anti-virus protection . Actually yesterday my Windows 10 had an huge update . Didn't checked out that some of the protection was enabled or not . But normally if there is something i have to recieve some kind of notification ? I mean it can be treated like a maware so Windows will detect it and let me know

McGiverGim, 2019-09-26:
If you haven't touch it, the anti ransomware is disabled by default (because it produces some problems similar to the one you are suffering), so is not the problem.

Searching at Google, it seems that maybe the file you have downloaded is corrupted. Can you try to download it again? You can download the 10.6-RC1 too to see if it works. Is stable enough to be used.

gmumdzhiev, 2019-09-26:
[https://github.com/betaflight/betaflight-configurator/releases/tag/10.5.1](url)
This is the only source that i found . From here i have downloaded my exe . Is there another one ?

McGiverGim, 2019-09-26:
No, but maybe something went wrong when you downloaded it the first time, please download it again:

https://github.com/betaflight/betaflight-configurator/releases/download/10.5.1/betaflight-configurator-installer_10.5.1_win32.exe

And you can test the 10.6-RC1 from here:

https://github.com/betaflight/betaflight-configurator/releases/download/10.6.0-RC1/betaflight-configurator-installer_10.6.0-RC1_win32.exe

gmumdzhiev, 2019-09-26:
I ran into the same errors on both of the exe's . I even tried it out on my laptop ( running the same new version of Windows 10 ) , but still the problem is there .

McGiverGim, 2019-09-26:
It seems some problem in your side, but I can say exactly which. We have thousands of installations and you are the first user with this problem.

If you see this link, for example (there are a lot of others similar): https://appuals.com/nsis-error-error-launching-installer/ there can be different causes:

1. The files for the software you have downloaded from the internet aren’t complete
2. The physical media (CD/DVD) you’re installing from is damaged
3. The downloaded files have been modified and are different from the original
4. Required hardware, such as the CD or DVD drive, is not working
5. You have a virus on your computer

gmumdzhiev, 2019-09-26:
I'am not here to blame , but how this should be from my side ? I mean i have tested it on my PC , my laptop and now on a colleague of mine ( that runs the older version of Windows 10 ) and the error is the same on 3 separate machines . I seropisly doubt  that the problem is coming from me .

McGiverGim, 2019-09-26:
Because several thousands users installed it since the release, in April, and nobody complained aobut it. 

I have downloaded it now and it works for me:
[image]

So is clear that you have something wrong in your network, computer, configuration, etc. but I don't know why.

mikeller, 2019-09-26:
@gmumdzhiev: This is not about blaming, this is about identifying where the problem is so it can be fixed. The thing is the binaries for both configurator 10.5.1 and 10.6.0-RC1 on the server are good - they've been downloaded by thousands of users and you are the first one to complain about this problem. So something must be corrupting these files on the way from the server to your computer. Can you try to download the file from a different computer, or even better from a different place, like your work or school?

gmumdzhiev, 2019-09-26:
@mikeller @McGiverGim  As i told in my previous posts . Yesterday i have downloaded it on my PC at home . And now i have tested it on my laptop at work and on a colleague at work . ( So seperate machines , different OS, different network ) . The error keeps poping out , so i don't know from where initially is coming the problem

gmumdzhiev, 2019-09-26:
I found the solution . I don't know why , but when i moved the installer.exe to another drive different from C:// it has launched without a problem .

closed 2019-09-30 by gmumdzhiev

mikeller, 2019-09-30:
@gmumdzhiev: Could it be that your C: drive is almost out of space?

frozenskys, 2019-09-30:
@mikeller seems to be a general solution to the "error launching installer" error from the nsis installer - most of the answers just seem to be "move the installer exe to another drive and it will work". I have had this before and that worked for me (I'm wondering if it has to do with windows 10 protected folders mechanism ?)

gmumdzhiev, 2019-09-30:
@mikeller I doubt it because i have around 150 GB free space at my C: drive . Actually I've saw so many videos on YouTube about your app guys and almost everyone is installing the installer.exe like this . Move it to the seperate drive and then install it . 
@frozenskys  actually may be right . It has something to do with the new windows secure folder mechanism . Just wont let you install third party programs if they are not secure enough