betaflight/betaflight-configurator #1720: Unable to install Configurator on MacOS (RC1 and RC2)

SuTeCH007, 2019-10-06:
"Betaflight Configurator.app" is corrupt and cannot be opened. You should remove the install file.

Chrome downloaded this disk image at 17:59 today from github.com.

[image]

I tried it multiple times on multiple access points. Nothing.

IvoFPV, 2019-10-06:
https://github.com/betaflight/betaflight-configurator/issues/1675#issuecomment-533765371 This seems to fix the issue for now. Could you try that?

SuTeCH007, 2019-10-06:
> [#1675 (comment)](https://github.com/betaflight/betaflight-configurator/issues/1675#issuecomment-533765371) This seems to fix the issue for now. Could you try that?

Yeah. That's it. Thanks. Now it's working.

closed 2019-10-06 by SuTeCH007