betaflight/betaflight-configurator #1721: won't on stall on macOS Mojave 10.14.6

S-T-A, 2019-10-06:
i installed the latest configurator 6.0rc1 no problem and i installed 6.0rc2 and it rplaced 6.0rc1 with rc2 and i get this message when i try to open it  “Betaflight Configurator 4” is damaged and can’t be opened. You should move it to the Trash. 
There is nothing wrong with my mac it was just formatted and it's clean i deleated it and reinstalled rc1 again no problem if there is something i'm doing wrong id like to know cheers

Hardware Overview:

  Model Name:	MacBook Air
  Model Identifier:	MacBookAir6,2
  Processor Name:	Intel Core i7
  Processor Speed:	1.7 GHz
  Number of Processors:	1
  Total Number of Cores:	2
  L2 Cache (per Core):	256 KB
  L3 Cache:	4 MB
  Hyper-Threading Technology:	Enabled
  Memory:	8 GB
  Boot ROM Version:	117.0.0.0.0

IvoFPV, 2019-10-06:
Please check other issues before opening yours. #1675 (comment) This seems to fix the issue for now. Could you try that?

mikeller, 2019-10-07:
Closing as duplicate of #1675.

closed 2019-10-07 by mikeller

S-T-A, 2019-10-07:
Alright cool I’ll give it a try thank you gentlemen 😎