betaflight/betaflight-configurator #1722: Add motor_output_limit to GUI

spuder, 2019-10-06:
It would be nice to be able to define the motor_output_limit in the gui, and not require the command line. 

https://www.youtube.com/watch?v=u2LMGDS7DFI

mikeller, 2019-10-07:
@McGiverGim: It is actually a PID Profile setting in the firmware. So it should go onto the PID tuning tab, and if it does not go into the PID profile subtab it will need at least a note stating that it is PID profile dependent.

Docteh, 2019-10-07:
I think `auto_profile_cell_count` goes together with `motor_output_limit`, so I think it makes sense to add them both at the same time.

the rate profile has throttle limit, so I think motor limit on pid tab makes more sense.

mikeller, 2019-10-08:
@Docteh: Yeah, the assignment of parameters to PID- and rateprofiles can be seen as 'organic' (at best). We should probably take a green field approach and define what should go where from scratch.

Juzzle1, 2020-03-03:
I think this is possibly more than a cosmetic issue

It's not obvious to users that the 'copy profile settings' button in the configurator overwrites the CLI-set `auto_profile_cell_count` and `motor_output_limit` on the destination profile.

While this makes sense as these settings are stored as part of the individual PID profiles, it could make it easy to smoke a motor, for example by copying a 4s profile to a 6s profile which was previously set to scale the motor outputs.

Maybe a warning to this effect would be a good idea when copying profiles if new GUI options aren't going to make it in the next release?

Asizon, 2020-03-03:
This feature is already added to configurator for 10.7.0
https://github.com/betaflight/betaflight-configurator/pull/1901

bolts1111, 2020-03-09:
I'm planning to have two quads running everything the same including motor size 2306. The difference being in motor kv one at 2450kv and the other at 2750kv. i'm then planning to run 6s and run the same rpm on each. Would the scaling work differently and feel different? Also how different is it running a true 6s / unscaled back motor?

Asizon, 2020-04-26:
Can be closed by https://github.com/betaflight/betaflight-configurator/pull/1901

mikeller, 2020-04-26:
Keeping this open for now - resizing of the camera frame is still something that could / should be done at some point.

Asizon, 2021-07-04:
Closed by #1901
@mikeller your resizing comment goes in https://github.com/betaflight/betaflight-configurator/issues/1833

closed 2021-07-04 by Asizon