betaflight/betaflight-configurator #1728: macOS 10.6.0 RC2 "is damaged and can't be opened"

ianmaddox, 2019-10-09:
When opening betaflight 10.6.0 RC2, I get an error from OSX stating "Betaflight Configurator.app is damaged and can't be opened. You should move it to the trash."

This happens immediately after pulling it out of the DMG file.

jp8080nl, 2019-10-09:
Can confirm this

IvoFPV, 2019-10-09:
https://github.com/betaflight/betaflight-configurator#note-for-macos-x-users
>Changes to the security model used in the latest versions of MacOS X 10.14 (Mojave) and 10.15 (Catalina) mean that the operating system will show an error message ('"Betaflight Configurator.app" is damaged and can’t be opened. You should move it to the Trash.') when trying to install the application. To work around this, run the following command in a terminal before installing: sudo spctl --master-disable. Then install Betaflight configurator, and after verifying that the installation has worked, run sudo spctl --master-enable.

ianmaddox, 2019-11-13:
IvoFPV's suggestion resolves this issue. Closing ticket.

closed 2019-11-13 by ianmaddox