betaflight/betaflight-configurator #1729: sudden ACC drifting / Matek F7 722SE / BF4.1.0-RC6?

hundemann, 2019-10-09:
Not sure if this is an issue, but I had on two same FC using that same FW a sudden ACC drift, after connecting to BF configurator (latest release).
What happened before: I was flying both already, but I almost had a nose crash with one of them during first launch. After connecting it to the configurator I noticed the ACC degrees were way off. 
The second one was flying great, but on that one I just could monitor now a continous drift, means while sitting on the table, the ACC was suddenly running off in the set-up screen.

On the first one I calibrated the ACC after I noticed the weird angles. And it is set to ~0 still now.
The second one just running off now I also calibrated the first time after FW flashing and it is also still on ~0°.

Both ones I did not calibrate after FW flashing, because I notices after connecting the ACC was calibrated perfectly (i mean really perfectly, usually after calibrating the ACC shows some 0,1 etc. values), so I did not bother further...

Thank you for any comment..
Malte

IvoFPV, 2019-10-09:
This often happens in flight due to vibrations. ACC will slowly drift because of that. Some ACCs are more prone to this type of issue. 
If you flash the older firmware it still happens, right?

hundemann, 2019-10-09:
No, never happened to me on these FC, on prior FWs I did not have this. I was quite careful to raise this as an issue, but this behaviour was never observed before.

The second drift did not happen during flight, it happened while the second quad was standing still in front of me,,,

IvoFPV, 2019-10-09:
@hundemann It could just be a defective ACC and you didn't notice it before. It is strange for it to drift while standing still. Never had that happen. 
Could you try flashing back to the previously used firmware version that you are sure you haven't had this happen and test it again? It is very unlikely for this to be a firmware issue.