betaflight/betaflight-configurator #1730: Dropping serial connection on linux, unable to reconnect

sanchosk, 2019-10-09:
I am using the RC2 version of the configurator (10.6.0) on Ubuntu.
When connecting to F4 (Mamba), everything works until I click save and reboot.
As soon as I do it, the board disconnects (expected), but does not reconnect (I have auto-connect on).
Even when I try to click the connect, I receive connection Failed error and in shell (where I start the betaflight-configurator) I see following error message:
[24876:24895:1009/223854.455217:ERROR:serial_io_handler.cc(149)] Failed to open serial port: FILE_ERROR_IN_USE

If I unplug and plug back the USB cable, the controller connects again no problem.
I had similar problem with previous version of betaflight-configurator on linux and what used to help was to change to different serial port and change back.
With this version, even this workaround does not work.

Perhaps I am doing something wrong here?

mikeller, 2019-10-10:
@sanchosk: I know it's not a fix, but I think the problem is auto-connect. From your description it sounds as if it is running into a race condition of the port still being open after the reboot.

Can you check if the problem still occurs if you disable auto-connect and connect manually after the reboot?

sanchosk, 2019-10-10:
I disabled auto-connect, did the same Save and reboot, after reboot beeps hit connect button, this is the result.
[Screenshot](http://sancho.sk/screenshots/betaflight-config-def1730.png)

mikeller, 2019-10-10:
Hmm, that's odd, configurator works for me just fine on Ubuntu 19.10. What version Ubuntu are you on?
Can you build a debug version and then check what is logged in the inspector console please: https://github.com/betaflight/betaflight-configurator#development

shanness, 2019-11-01:
I'm having the same issues after upgrading to 10.6.0 (although haven't run BF for about 6 months).  If I keep trying the connect button eventually it works..

I also have auto connect disabled
[6278:6285:1101/234451.355227:ERROR:serial_io_handler.cc(149)] Failed to open serial port: FILE_ERROR_IN_USE

closed 2019-12-08 by stale[bot]

genna87, 2020-02-03:
As hinted [HERE ](https://github.com/betaflight/betaflight/wiki/Installing-Betaflight#platform-specific-linux) the problem is ModemManager that thinks the FC is a GSM modem

I had the very same problem and I solved uninstalling it since I didn't need it:
`sudo apt remove modemmanager`