betaflight/betaflight-configurator #1731: Tooltip for RPM data occasionally remains popped up indefinitely

jflyper, 2019-10-10:
10.6.0-RC1
Happens very frequently (but not always).
When it happens, there's no way to make it disappear other than quitting the app.

[image]

IvoFPV, 2019-10-10:
Get the 10.6.0-RC2 version, it has been fixed.

McGiverGim, 2019-10-10:
Yes. This happens with others tooltips too, but the RPM one is more frequent.

We have tried with several changes here https://github.com/betaflight/betaflight-configurator/pull/1604 and here https://github.com/betaflight/betaflight-configurator/pull/1665 it seems without luck :(

jflyper, 2019-10-10:
Hmm. RPM tooltip seems to be fixed with RC2.

(Btw, the RPM tooltip text should be "These numbers" instead of "This number", right?)

mikeller, 2019-10-10:
> "These numbers" instead of "This number"

We definitely need add a translation from 'developer' to English. :wink:

closed 2019-11-16 by stale[bot]