betaflight/betaflight-configurator #1744: VTX table - how to input only one band?

laurentopia, 2019-10-13:
I only use A4, how do I input that in the table?
[image]
this doesn't work, the vtx remains at R1

Anisotropic, 2019-10-14:
You configured A band with 1 channels with frequency 4. If you want A4 = you need to input 4 channels per band and in 4 channel (last) you need to put your frequency

IvoFPV, 2019-10-14:
You can just set the frequency of A4 channel instead of 4. Also your power values are wrong.
Please don't open an issue regarding this. This is a configuration issue and not a right place to get help. You can join our Slack where you can ask these kind of questions and get the answer much quicker. https://slack.betaflight.com/

mikeller, 2019-10-15:
@laurentopia: Try this: 
[vtxtable.json.txt](https://github.com/betaflight/betaflight-configurator/files/3728668/vtxtable.json.txt)

laurentopia, 2019-11-15:
> @laurentopia: Try this:
> [vtxtable.json.txt](https://github.com/betaflight/betaflight-configurator/files/3728668/vtxtable.json.txt)

cool, this helps and so this should be the default.

closed 2019-12-22 by stale[bot]