betaflight/betaflight-configurator #1745: VTX Tab Freezing

Phiqtional, 2019-10-14:
Hi Guys,

So I'm currently running the 10.6.0 RC2 configurator and when I select the Video transmitter tab, it freezes. Is there anyone else experiencing this? The same thing happened when I installed the new 10.6.0 alpha configurator as well.

Once I edit the VTX table, it won't allow me to save either.

[image]

IvoFPV, 2019-10-14:
@Phiqtional Get the official release version 10.6.0 of the configurator. Make sure you are running the latest firmware version 4.1-RC6 as well.

Phiqtional, 2019-10-14:
I've already downloaded the latest configurator, AND also running latest firmware 4.1 RC6 but still experiencing this. Not sure why though.

IvoFPV, 2019-10-14:
In the screenshot you have 10.6.0-RC2 that's why I asked. Could you provide a `diff`? Have you tried reflashing with full chip erase enabled? Can you try without the battery plugged in?

Phiqtional, 2019-10-14:
After reflashing again, the vtx tab now works. This was so odd to experience. Thanks for the response!

McGiverGim, 2019-10-15:
It is fixed, so I close it.

closed 2019-10-15 by McGiverGim