betaflight/betaflight-configurator #1748: Windows 10: Configurator Solid Black Screen

kgustine, 2019-10-17:
Running Windows 10 Version 1903 (OS Build 18362.418) w NVIDIA Quadro 1000m video card.

I just installed 10.6.0. When I launch it, the screen is black with a white square in the middle. After I resize the window, the entire contents are solid black. If I move my mouse around, the cursor changes and I was able to open the website by apparently clicking some link that I could not see.

I did reboot, which made no difference

[image]

IvoFPV, 2019-10-17:
Could you try reinstalling the Configurator and updating your GPU drivers?

mikeller, 2019-10-17:
@kgustine: Looks like a GPU issue. Can you try if starting the application with the `--disable-gpu` option fixes this?

kgustine, 2019-10-22:
I had already tried reinstalling it, but launching it with --disable-gpu allows it to display correctly

mikeller, 2019-10-22:
@kgustine: Sounds like this is solved then. Can you close please?

closed 2019-10-22 by kgustine

kgustine, 2019-10-24:
This seems more like a band-aid. Is this a known issue, and is there an idea of how to fix it?

Docteh, 2019-10-24:
It's a known issue that a few people see, but I don't think anyone here is
interested in making builds of nwjs. Nwjs is the item that uses the chrome
browser code base for stand alone programs. I think to properly fix it we
might have to determine what chrome does that nwjs is not doing.

If I had access to a computer that exhibits black screen I'd be tempted to
try a few things, see if anything works. Updating nwjs might make things
worse instead of better. https://github.com/nwjs/nw.js/issues/7116 has a
few reports of black screen on Windows 7 with a newer version of nwjs than
we shipped 10.6.0 with.

mikeller, 2019-10-24:
@kgustine: Yes it is a band-aid that we have to apply to work around something that is broken upstream in the NW.js toolkit. So if you want to take this any further, I suggest that you report this bug on https://github.com/nwjs.