betaflight/betaflight-configurator #1751: Suppress manufacturer ID in target list when unnecessary

etracer65, 2019-10-17:
Having the manufacturer string in the target names is causing a lot of confusion and support questions. I propose suppressing the manufacturer ID **unless it is truly needed to prevent ambiguity for a given target**. I realize it's there to potentially allow the case where multiple manufacturers might use the same target name. But we currently have no occurrences of this and in general it's an unlikely scenario as the manufactures would want to avoid the branding confusion (I'd assume).

mikeller, 2019-10-18:
Agreed. The current solution was a workaround to solve the issue of duplicate target names in the repository that is feeding into configurator.
The solution in configurator was not the best possible approach, as it was done last minute.
I think in order to get the overall best user experience with what I expect to be a rapid increase in target numbers (as boards that are currently re-using existing designs like OMNIBUSF4 will want to add their own target configurations), a solution will have to be implemented that allows users to browse by manufacturer **and** by target name.
But as an interim, if this is not going to be ready by the next configurator release, a solution that omits manufacturer ids (which aren't very intuitive for users anyway) for targets that do not have duplicated names.

mikeller, 2019-11-17:
Fixed by #1820

closed 2019-12-17 by etracer65