betaflight/betaflight-configurator #1752: Betaflight 10.6 app won't open on mac

gzhun, 2019-10-18:
I just installed the Betaflight 10.6 on my mac and won't open the app. it says "Betaflight Configurator.app is damaged and can’t be opened. You should move it to the Trash". I did went to the Terminal and put in this line "sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app". but still the same. can't open the app. any idea what could be the problem? (I kept the old version on my mac -so there is two Betaflight in the Apps now- for the older kwads but it can't be the problem IMO)

gzhun, 2019-10-18:
this vid is the key how to open that app (or any app):  https://www.youtube.com/watch?v=_gQA3RhcJuU

closed 2019-10-18 by gzhun

mikeller, 2019-10-19:
@gzhun: What is shown in this video is not an ideal solution, and will not work in every case. It is preferrable to use the solution that is suggested in the [release notes](https://github.com/betaflight/betaflight-configurator/releases/tag/10.6.0):

> Changes to the security model used in the latest versions of MacOS X 10.14 (Mojave) and 10.15 (Catalina) mean that the operating system will show an error message ('"Betaflight Configurator.app" is damaged and can’t be opened. You should move it to the Trash.') when trying to install the application. To work around this, run the following command in a terminal before installing: `sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app`.

funk3h, 2019-10-24:
doesn't work for me. terminal asks for a password

wind0r, 2019-10-24:
and what happens when you fill in the correct password for your admin user?

gzhun, 2019-10-24:
> doesn't work for me. terminal asks for a password

than type the password into terminal and hit enter

gzhun, 2019-10-24:
> and what happens when you fill in the correct password for your admin user?

you gonna have BF 4.1 working

funk3h, 2019-10-24:
ok got my password working, but still won't install despite putting in the command for the terminal keeps telling me its corrupt and move to trash

funk3h, 2019-10-24:
sorry just saw your last reply, should I be trying  4.1 instead of 10.6.0?

gzhun, 2019-10-24:
> ok got my password working, but still won't install despite putting in the command for the terminal keeps telling me its corrupt and move to trash

did you followed this tutorial? https://www.youtube.com/watch?v=_gQA3RhcJuU

funk3h, 2019-10-24:
no, will give it a go, cheers

kardady, 2020-04-15:
> doesn't work for me. terminal asks for a password

have you solve your problem? mine ask a password too

kboardm, 2020-05-09:
Solved the password. But still doesn’t load even with the sudo command. Any ideas

AndrePaco, 2021-05-08:
hallo ich habe mal eine frage ich habe ein ipad 12.9 und möchte gerne betaflight für mein freund drauf spielen habe es runter geladen aber kann es nicht öffnen und ich habe keine ahnung was ich noch machen soll 
ich muss noch dazu sagen das ich noch nie ein pad hatte mit ios software und ich absolut keine ahnung habe was ein terminal ist oder sonst was
lg anna