betaflight/betaflight-configurator #1762: Receiver tab throwing Javascript errors if receiver not configured

etracer65, 2019-10-21:
About once a second the follow error is generated:
```
Error in event handler for serial.onReceive: TypeError: Cannot read property 'length' of undefined
    at update_ui (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/tabs/receiver.js:479:40)
    at MspHelper.process_data (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp/MSPHelper.js:1558:31)
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp.js:139:13
    at Object.notify (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp.js:138:24)
    at Object.read (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/msp.js:126:26)
    at read_serial (chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/serial_backend.js:480:13)
```
Configuring a receiver type stops the errors. Tested with Configurator current master and with firmware 4.1.

McGiverGim, 2019-10-21:
It seems to fail here: 

https://github.com/betaflight/betaflight-configurator/blob/86901b931809bdee0233cc57fd9c70aee415dc63/src/js/tabs/receiver.js#L479

Because the array has no elements when there is not active channels.
If someone want to fix it, perfect, if not I will do it in the next days.

McGiverGim, 2019-10-29:
@etracer65 I'm looking at this but I can't reproduce it. Can you post a diff or something similar? Maybe I need to enable some option to make this problem happen.

etracer65, 2019-10-29:
Just make sure that no UART is selected for Serial Rx on the Ports tab, and that serial receiver is selected for the type on the Configuration tab. Then whenever in the Receiver tab it will through the error once a second.

closed 2019-11-02 by mikeller