betaflight/betaflight-configurator #1766: Error in deb package

andreasschramel, 2019-10-23:
Unable to install betaflight-configurator_10.6.0_amd64.deb package under Ubuntu 19.10. 
The following error messages appear:
<pre><font color="#3465A4"><b>~/Downloads</b></font>$ sudo apt install ./betaflight-configurator_10.6.0_amd64.deb 
[sudo] password for andreas: 
Reading package lists... Error!
E: Problem parsing dependency 21 of betaflight-configurator:amd64=10.6.0
E: Error occurred while processing betaflight-configurator (NewVersion2)
E: Problem with MergeList /home/andreas/Downloads/betaflight-configurator_10.6.0_amd64.deb
E: The package lists or status file could not be parsed or opened.</pre>

mikeller, 2019-10-23:
@andreasschramel: Try downloading the file again - AWS was having issues today, so it's possible that the file you downloaded was corrupted.

andreasschramel, 2019-10-23:
@mikeller did not solve the problem, I still have the same output on the terminal.

mikeller, 2019-10-27:
@andreasschramel: Wait a second - apt is intended to be used to install packages that are published through a repository, which betaflight configurator isn't.

Try this instead:

```
sudo dpkg -i ./betaflight-configurator_10.6.0_amd64.deb
```

closed 2019-12-03 by stale[bot]