betaflight/betaflight-configurator #1772: 10.7.0 - does not flash firmware anymore

TheIsotopes, 2019-10-27:
bfc 10.7.0 does not flash anymore since #1763  was merged 

@mikeller can you check what's wrong

[image]

mikeller, 2019-10-27:
@TheIsotopes: Can you supply a bit more information on what operating system this is, what board you were connected to, and what firmware you were trying to flash?
I was using Betaflight configurator 10.7.0 built from latest master to flash a BEEROTORF4 with a STM32F405 target on linux just now, and it worked just fine.

TheIsotopes, 2019-10-27:
@mikeller sry ... using Windows 10 (Version 1909 Build 18363.449)
no flashing with all my fc and the firmware does not matter

mikeller, 2019-10-27:
Erg, looks like this the version of chrome that is bundled into 0.41.3 of the NW.js toolkit is broken in Windows:

```
reporting chrome error: The device is not in a configured state.
stm32usbdfu.js:107
```

Will need to investigate - for now use the released version of configurator.

PR #1794 (2019-11-03): Update version of nw.js
this one fixes #1772
closed 2019-11-03 by mikeller