betaflight/betaflight-configurator #1778: Warning to update to 10.6.0 configurator when already using it.

SunshynRFF, 2019-10-28:
When I click on Betaflight and it loads, I get a Warning message that there is a newer version of configurator and mine is outdated.  Has link to release page, which is where I downloaded and updated to 10.6.0 a month ago.  When I close warning, BF does not give me anymore notifications, or run any differently.  Only when I close BF completely out and re-open it do I get the warning.

[image]
[image]

mikeller, 2019-10-28:
@SunshynRFF: You are still using the 10.6.0-RC1 version of configurator:

[image]

Try uninstalling configurator first, and then installing the 10.6.0 version.

SunshynRFF, 2019-11-03:
Oh ok, I can do that, thank you!!

Jack ley
Sunshynff@yahoo.com
636-262-2152

closed 2019-12-10 by stale[bot]