betaflight/betaflight-configurator #1780: Using localization before ready

McGiverGim, 2019-10-29:
When starting the Configurator, we have a lot of error similar to this one:
```
i18next.min.js:1 i18next: hasLoadedNamespace: i18next was not initialized Array(2)
output @ i18next.min.js:1
i18next.min.js:1 i18next::translator: key "windowTitle.message" for namespace "messages" won't get resolved as namespace was not yet loaded This means something IS WRONG in your application setup. You access the t function before i18next.init / i18next.loadNamespace / i18next.changeLanguage was done. Wait for the callback or Promise to resolve before accessing it!!!
```
This happens because:
- We ask to load the localization system, but this happens in async way
- We load all the js in the main page
- A lot of js use the localization system. It is not loaded at this moment, so errors are thrown
- The localization system really loads it now

I can find an easy way to fix this. There are some sync backends for the i18next library, but we can't use it while we have a Chrome app version.
I open this as an issue because maybe someone has an idea to fix it.
At this moment it does not seem to produce problems in the execution of the Configurator, but maybe at some moment it does.

Docteh, 2019-10-29:
I was wondering about that error. What sort of sync backends are problematic? We might be able to repackage via webpack or browserify

Docteh, 2019-10-29:
I found this issue on i18next https://github.com/i18next/react-i18next/issues/977

For our 10.6.0 we shipped with version `17.0.11` of i18next, but more recently the use of  "^17.0.11" in `package.json` means that we got automatically updated to `17.3.0`. 

I downgraded to `17.0.11` and the errors go away. Updating to `18.0.1` as mentioned in the issue also makes the errors go away. 

Interestingly enough they also have a `19.0.0` version posted.

mikeller, 2019-10-29:
Is this actually an indication of something being broken, or is this just a message that can be ignored because everything works as it should? #1782 is possibly all that we need.

McGiverGim, 2019-10-31:
Fixed in https://github.com/betaflight/betaflight-configurator/pull/1782

I close this.

closed 2019-10-31 by McGiverGim