betaflight/betaflight-configurator #1785: [Feature Request] Add KV to RPM in Motor tab

JyeSmith, 2019-10-30:
I know it can easily be calculated but it would be super convenient to have it right there,  Great for checking rewound motors, or specs on new motors.

Cheers

McGiverGim, 2019-10-31:
The only problem with this, is that the space in the motor tab where we put the telemetry values, is very reduced. We can add one or two more values, but I don't think we can add any more. If more people agree that this is a valuable addition, I can add it.

McGiverGim, 2019-11-12:
I have started to implement this, but I have found a problem: to calculate the KV, I need the voltage that is using this motor... the RPM bidir protocol does not provide this information. The old ESC SENSOR provides it, but you need the additional wire.

Am I missing something? Or do you want to provide it only when the additional wire is connected? Will be use the general voltage as a value for all the motors enough precision for this?

JyeSmith, 2019-11-12:
You should be able to use vbat.

AlessandroAU, 2019-11-12:
Yerp, kV = RPM / (vbat * commanded throttle)

McGiverGim, 2019-11-12:
I need to use the commanded throttle? I have started with it, better to continue the discussion in the PR https://github.com/betaflight/betaflight-configurator/pull/1823

haslinghuis, 2023-06-21:
@McGiverGim @JyeSmith @damosvil thoughts?

damosvil, 2023-06-21:
> I have started to implement this, but I have found a problem: to calculate the KV, I need the voltage that is using this motor... the RPM bidir protocol does not provide this information. The old [...]

EDT provides voltage in the Voltage frame, always the esc senses the Voltage and implements the frame. I remember that Blheli_32 implements It in an esc I have in a iF Titan. 
The EDT is documented here https://github.com/bird-sanctuary/extended-dshot-telemetry
Betaflight already implements it since Betaflight 4.4, and it is integrated in dshot.c

haslinghuis, 2023-06-22:
Can be closed.

closed 2023-06-22 by haslinghuis