betaflight/betaflight-configurator #1816: Suggestion: present warning when max7456 OSD chip not available

etracer65, 2019-11-09:
Currently if the max7456 is not available the div on the OSD tab that holds the "Font Manager" button is hidden. For reference:

https://github.com/betaflight/betaflight-configurator/blob/a656dc2f5bbd43f1dd9c3292f8b25e69f6968217/src/js/tabs/osd.js#L2365-L2367

Previously the controlling flag was simply based on whether or not the target was defined to use the max7456. Now there's an additional layer where the max7456 needs to be actually detected (communicating) before this flag is active.

So it would be a good idea to present some information to the user if the font manager division is hidden as now it's possible for some flight controllers to not have the max7456 detected when only powered via USB (see https://github.com/betaflight/betaflight/pull/9158). Otherwise users might be confused when the "Font Manager" button is not visible. Maybe a message like:

> No OSD chip was detected. Some flight controllers will not properly power the OSD chip unless connected to battery power.

Maybe change the logic to not hide the division, but instead make it disabled and add the informational message.

mikeller, 2019-11-09:
Good idea!

etracer65, 2020-03-19:
This has come up a few times now that people have started to use the 10.7 Configurator nightlies so we might want to address this to minimize support issues after the release. It's caused confusion about why the font manager button is missing when they have a board that doesn't power the max7456 from USB. While it's correct that the button is missing (since it can't work), people are confused because they thought it was working before. Previously the font would appear to upload (progress bar moved) but it didn't actually work.

Maybe an alternative to hiding the button entirely would be to display a "disabled" version of the button with a hover message explaining that the OSD chip wasn't detected.

mikeller, 2020-03-19:
Good point @etracer65. Hover on a disabled element is missed by some users - maybe change the text on the disabled button to 'OSD chip not detected'?

Asizon, 2020-04-26:
Can be closed by https://github.com/betaflight/betaflight-configurator/pull/1927

closed 2020-04-26 by mikeller