betaflight/betaflight-configurator #1828: MacBook Pro won't detect drone

FreshCoastDon, 2019-11-15:
Hello,

This is my first post, so easy on the roasting please and thank you. Anyways, I was able to install Beta Flight Configurator 10.6.0 and the CP210X drivers fairly easy. When I have Beta Flight up and connect my drone via usb c cable, nothing happens. I have tried it as auto-connect and manually pushing connect to no avail. Any ideas on what may be causing this connection issue? I have also verified the cables, all are good. I have also tried a LiPo battery attached and not attached while connecting and nada. I can provide screen shots if you have any input as well.

b3rgman, 2019-11-17:
I am having the same problem.  Do I need to specify a certain com port?

FreshCoastDon, 2019-11-17:
I'm not sure myself... I'm going to try it on my work laptop and go from there until I can find a solution. My drone is kind of sitting here and I'm itching to fly it.

Anyone else in the community care to chime in? Thanks in advance...

b3rgman, 2019-11-18:
I switched to Mac recently. It used to work for me on Windows.   I made sure the firewall has allowed the connections from BetaFlight as well as the CP210 driver.  I cant get it to pick up the driver.   My guess is its Catalina and I am not being promted for something.

jokster945, 2019-11-22:
I'm having the exact same problem. Anyone have a fix yet?

etracer65, 2019-11-26:
For modern flight controllers you don't need to install any drivers on MacOS. Try using a different USB cable - some are only charging cables and don't have the data lines connected.

FreshCoastDon, 2019-12-27:
hey guys, I was finally able to connect my drone to my MAC. Thanks for the assists and tips here.

closed 2019-12-27 by FreshCoastDon

iBreaker, 2020-01-04:
@FreshCoastDon  hi, I have the same problem. How to  fix this.?  thanks.

OSX 10.15.2

kholder116, 2020-07-21:
> hey guys, I was finally able to connect my drone to my MAC. Thanks for the assists and tips here.

can you please tell us how you were finally able to connect to your MAC.... I'm ready to throw this drone out the window...   I feel as though I should never have upgraded to Betaflight 01.7.0, it was working perfectly before ..

kholder116, 2020-07-21:
Has anyone else had this c- cable MAC not connecting to Betaflight issue and finally resolved it. If so , could you post what you did. 
 thanks.

Brocknelson12, 2020-11-06:
Anyone had any success? It won't show up on my usb devices on my Mac and I can't connect through beta flight either

etracer65, 2020-11-06:
@Brocknelson12 Works just fine. No problems connecting when using Mac. Most of the time the problem people have is a bad cable. You must use a real USB data cable and not just a charging cable like which came with your phone. If you try another cable and it still doesn't work, then try more cables.

Brocknelson12, 2020-11-07:
Okay i tried with a cable that ive used to write to other devices(tinyduino) so i know its a data cable. I have mac OS Catalina 10.15.7. And DYS f4 pro v2. I have read that the new Driverkit only supports CP2102(for SILabs) and i am not sure that my flight controller uses that or not.

mikeller, 2020-11-07:
@Brocknelson12: No, the DYSF4PRO uses VCP. This works just fine with MacOS Catalina if the correct cable is used.

skywinder, 2020-11-15:
same issue on 10.7.0 have no idea how to fix..

GerrieWell, 2021-08-22:
I was blocked by this too. My drone is mobula6, just connect it to my mac with many cable. The system repport doesn't show any new device. And I haven't any window device to try. The mobula6 is sucks!

markjaquith, 2022-01-19:
I thought my USB-C to Micro USB adapter passed data, but I was wrong. So even if you've tried multiple cables, try something else. In the end, what worked was a USB-C to USB-A adapter, and then a USB-A to Micro USB cable.

Revo-Lucien, 2022-12-27:
Tacking on the importance of making sure you have the right cable, I got the Foxeer f722 v4 which takes a USB-A. When I plugged the FC in the correct lights flashed on and it looked ready to go, but sure enough the mac wasn't detecting anything. After a lot of frustration I went to a shop and bought a brand new USB(A->C) cable and... it worked.

christhomas, 2023-07-13:
sometimes when I flash betaflight my drone isn't detected anymore, so I have to downgrade it and I guess the problems get fixed up when more people notice, so try downgrading your version too. That sometimes helps

SebiUNG, 2023-09-01:
Same for me, the USB cable was not passing data.

nerdCopter, 2023-09-01:
any new computers that use USB-C only connections might now need "OTG" cables or USB-A to C adapters. just passing along what i've read.  i have no mac experience.  side-note: An adapter does work for me on a dell XPS with USB-C only, whereas a double-ended-usb-c did not.

JulianSchmid, 2024-07-25:
It works for me only if I chain USB-C to A to C again with adapters. USB-C directly did not work.

SDFfpv, 2024-10-20:
I used an adapter that I thought had data transfer but didn't. I would plug your drone straight into your computer.

dev-mansonthomas, 2025-06-07:
My google search bring me here for the same issue, and being a beginner. Noticing that I had the same issue on my PC, I felt the issue was elsewhere. And I also noticed the microusb port in addition to the more visible USB-C port on my drone.

I'm using "NewBeeDrone AcroBee75 Pusher DJI O4 HD ELRS 2.4GHz 2S" as a first drone.

The drone has 2 ports : 
* on top of the drone, next to the batterie strap, at the rear of the drone, there's a MicroUSB port : **this is the correct drone**

[image]

* under the drone, an USB-C port : this one is I guess for the video module and it is **not** the correct port

[image]

Once connected, you need to change the device selection (above the auto-connect button) to something like /dev/tty.usbmodemxxxxxx

[image]