betaflight/betaflight-configurator #1829: Betaflight in wrong language

jasqsr11919, 2019-11-28:
I followed Joshua Bardwell's link to this betaflight configurator and it installs the configurator in Indonesian for some reason. I am unable to figure out how to switch it to english. Thanks for the help!

mikeller, 2019-11-28:
@jasqsr11919: Maybe you have to ask @joshuabardwell how to fix this in the version he is linking to then?

The currently released version of configurator has the language selector at the bottom of the startup page: 
[image]

jasqsr11919, 2019-11-28:
oh wow ive been looking for this option since the release. it was weird. I downloaded every configurator before this and they all opened in english except this one. Thank you!

closed 2019-11-28 by jasqsr11919