betaflight/betaflight-configurator #1833: Add support for camera frame OSD element

etracer65, 2019-12-06:
Recording this so we don't forget to add support for https://github.com/betaflight/betaflight/pull/9261

Needs to be a little more than a simple element. The preview will have to take into account the width/height that are supplied separately in the `MSP_OSD_CONFIG` message so that the "frame" can be rendered properly. Ideally we should also let the user modify the width/height and these are supplied back the flight controller in `MSP_SET_OSD_CONFIG`.

My ideal scenario is to add "resize handles" to the corners and let the user drag and resize the camera frame. This would then update the position and width/height as appropriate.

This is potentially an issue with large elements wrapping in the preview that may still need to be addressed from this incomplete PR: https://github.com/betaflight/betaflight-configurator/pull/1324

Asizon, 2020-04-26:
Can be closed by https://github.com/betaflight/betaflight-configurator/pull/1879

Asizon, 2020-04-26:
Tru @mikeller, we still need to enable frame resizing

haslinghuis, 2024-05-06:
Fixed in #1879

haslinghuis, 2024-05-06:
> My ideal scenario is to add "resize handles" to the corners and let the user drag and resize the camera frame. This would then update the position and width/height as appropriate.

Reopening to add a mechanism for 

The normal OSD element position is used for the top-left of the frame. New parameters are available for the width/height.
```
osd_camera_frame_width
osd_camera_frame_height
```
The width/height are added to the MSP_SET_CONFIG and MSP_SET_OSD_CONFIG MSP messages to support configuring the frame size in the Configurator.