betaflight/betaflight-configurator #1834: Installer Hangs on Uninstaller

tlebrell, 2019-12-09:
I can't get the 10.6.0 configurator installer to progress past the uninstaller. System is Win10 Home 64 on an i5 gaming box with 16GB RAM. I also ran as admin. I'm attempting to install onto the D: drive (mechanical) instead of the OS drive to save space on the SSD.

[image]

Asizon, 2019-12-09:
Just wait for finishing, now de installation take some minuts.

McGiverGim, 2019-12-09:
You have installed it directly in program files, without subfolder, I suppose by error. It is listing ALL the files from this folder for the uninstalling script.

mikeller, 2019-12-09:
Ouch, that is going to be a while...

closed 2020-01-15 by stale[bot]

Gordo5556, 2021-04-30:
I am experiencing the same issue currently. Has anyone found a solution to this?

piotrva, 2021-11-05:
Same issue for me - hangs on same step and I am installing in default location. 
Windows 10
Tested on two separate machines - same effect...
[image]

580, 2021-11-06:
Same issue here, Windows 11:
[image]

Left it for around 15 minutes, and installation completed.

Cronople, 2021-11-09:
[image]

Same issue, window 10 Edu, 20H2 ver.

I'm trying to install from the same folder where betaflight is installed.

over 30 min. still installing...

flyTib, 2021-11-16:
[image]
same here

Rubines, 2021-12-01:
same here insane :)

gavnunns, 2021-12-10:
This problem still exists and should be reopened.  I have tried multiple windows machines and the result is the same, hangs indefinitely creating the uninstaller.

Asizon, 2021-12-10:
@gavnunns did you try development 10.8.0 build? Is here where the issues are adressed.

gavnunns, 2021-12-10:
@Asizon I was using 10.7.1
I will give 10.8.0 dev build a go later today.
Thanks for the suggestion ;)