betaflight/betaflight-configurator #1838: Vide Transmiter tab freezes

kzurada, 2019-12-15:
Hi. The Video Transmiter freezes if you put number of power levels > 0 without labels on them. Then the tab shows very strange values with no save button and so the whole configutar freezes too. This applies to version of beta flight 4.1.1 with configurator 10.6

McGiverGim, 2019-12-21:
I can't reproduce it with latest developer versions. I think remember some fixes were added to the Configurator since the release. Can you test the latest developer version to see if it is fixed? You can find it here: https://github.com/betaflight/betaflight-configurator-nightlies/releases

szafranski, 2020-01-01:
I had the same issue. I even made a screen shot. Only way to get rid of that was to manually add vtx table by CLI, something like:

vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

saving, and then tab started to act as it should. 

[image]

McGiverGim, 2020-01-01:
Again, if you can, test the actual developer version to see if it is fixed. I can't reproduce it with the developer version.

szafranski, 2020-01-01:
Issue still present in 10.7 RC

[image]

Asizon, 2020-01-01:
Unable to reproduce under Windows 10 system, all Video Transmitter tab work as expected.
[image]

szafranski, 2020-01-01:
Maybe this will help:

https://youtu.be/1q4aALY4gkU

McGiverGim, 2020-01-02:
@szafranski I have reproduced your bug... is not because the power level tal is empty, is because is empty and you have it selected as actual power level in the upper part.

I will look into it for a fix...

szafranski, 2020-01-02:
Ok, glad I helped. Cheers

closed 2020-01-16 by mikeller