betaflight/betaflight-configurator #1842: Linting / Code Formatting

yngndrw, 2019-12-23:
Currently there is little to provide a definition of the coding standards other than the pre-existing code (Which isn't always consistent) and the [SonarCloud Code Analysis](https://sonarcloud.io/dashboard?id=betaflight_betaflight-configurator).

I've already raised a PR (#1841) to add a basic [EditorConfig](https://editorconfig.org/) file which defines the core IDE settings. (Indent style and indent size)

A [linter](https://eslint.org/) can be used to detect and fix bad practices much in the same way as SonarCloud does, but the ESLint configuration file can be used to control plugins for a [variety of IDEs](https://eslint.org/docs/user-guide/integrations). (I personally use VSCode and it works very well in that.) I wouldn't personally add it to the build as SonarCloud takes care of that, I'd only suggest adding the [`.eslintrc` configuration file](https://eslint.org/docs/user-guide/configuring) to the repository in order to drive ESLint contributors if they have the plugin.

A code formatter such as [Prettier](https://prettier.io/docs/en/index.html) works [hand-in-hand with the linter](https://prettier.io/docs/en/comparison.html) to provide automatic enforcing of formatting rules. This has the advantage of removing any ambiguous formatting decisions such as how to layout multi-line code, which most of the time comes down to little more than personal preference. As with ESLint, it is [configured by a `.prettierrc` file](https://prettier.io/docs/en/configuration.html) and has [integration with the most common IDEs](https://prettier.io/docs/en/editors.html). The IDE plugins often offer an optional "format on save" mode which is really nice in use.

mikeller, 2019-12-23:
It's not very obvious, but we actually do have coding style rules: https://github.com/betaflight/betaflight/blob/master/docs/development/CodingStyle.md

This is obviously not always 100% applicable as it is targeting C and not JavaScript, but since most of the core contributors are active in the firmware and configurator repositories it seems to be reasonable to stipulate that the coding style be the same in both where applicable.

yngndrw, 2019-12-24:
It never even occurred to me to check the main BetaFlight repository. If that should be applied here, it should definitely be copied over into this repository.

The problem with trying to keep coding styles consistent across C and JS is that you'll end up with JS styles which differ from the conventions that are typically applied to JS and this may end up alienating would-be contributors who are used to JS.

An example of a JS convention would be that braces are nearly always placed on the same line which [differs from the style shown in the BetaFlight repository](https://github.com/betaflight/betaflight/blob/master/docs/development/CodingStyle.md#curly-braces). This particular rule is especially prevalent in the JS community and even seems to apply to the existing codebase for the configurator, despite either style being technically possible and usable.

Speaking of that particular rule, the opening brace and closing brace rules in the BetaFlight repository seem to be mutually exclusive.

There are also a number of JS specific rules that you'd really want to define, such as not using `var`.

mikeller, 2019-12-24:
@yngndrw:

> It never even occurred to me to check the main BetaFlight repository. If that should be applied here, it should definitely be copied over into this repository.

Copying is a bad idea in my opinion - it will just open the door to inconsistencies being introduced. Also, since configurator and the other repositories under https://github.com/betaflight/ are really just 'helper applications' for the firmware, and do not have much (if any) value without it, we are generally trying to keep information like https://github.com/betaflight/betaflight/tree/master/docs/ and the wiki concentrated in one place, and that place is the firmware repository. Again, spreading out over different locations what is really just facets of the same matter won't improve things.

> The problem with trying to keep coding styles consistent across C and JS is that you'll end up with JS styles which differ from the conventions that are typically applied to JS and this may end up alienating would-be contributors who are used to JS.

The alternative is to alienate core developers and contributors to the firmware, many of whom are rooted in the C world, making it less likely that configurator support for new firmware features is contributed by the contributor of the feature - not sure which option is better.
But I am not saying that we should not try to come up with rules specific to the JavaScript projects, that are based on the 'master rules' for the firmware, but define variations made for JavaScript, and extensions for syntax that isn't present in C.

closed 2020-01-31 by stale[bot]