betaflight/betaflight-configurator #1847: Flight Session Automation

simonbethke, 2019-12-28:
I have a feature request that I might implement myself if people who are familar with the sourcecode could help me getting at a good starting point.

User Story:
As a user of betaflight, I want to spend less time with processing blackbox log files during my flight sessions.

Background:
In a flight session I usually only have the trunk of my car to manage copters, batteries, goggles, remote, antennas, tools... And for tuning I additionally need a notebook. Now, to make use of the blackbox logs, I usually need many iterations flying, downloading the log, running it through plasmatree, checking it in bbl explorer, adjusting the copter and fly again. This is very tideous espectially with the small amount of space, maybe cold temperatures outside and maybe even no space for using a computer-mouse.

Idea/Details:
The betaflight configurator should get a new page that can be accessed before connecting a copter. It should be in the menu below the firmware-flasher.
This tool can be set up to connect to a copter automatically and do a set of tasks. My personal goal would be to have it download all blackbox log files to a certain folder and rename them to something containing the craft-name, date-time and maybe some session-name. Then it should also clear the blackbox-storage on the copter and run plasmatree on the files. After the blackbox is cleared and at the same time plasmatree is doing the calculations, the configurator could already open and just to the pid-tab.

Implementation:
I would go for a very basic and easy solution first. I would consider predefined steps that can be configured slightly. Maybe when enabling this, a modal window would stay open and provide status information when connecting a copter. So it is also granted that no other controls interupt the process.

mikeller, 2019-12-28:
Not sure that this should be integrated into the configurator - having one tool for one job only is normally the most versatile, and most maintainable approach.
This is especially true as the MSC emulation, and downloading of log files through the operating system's file management is the fastest, most reliable, and recommended way to get log files off the flight controller these days.
My approach for this would be to create a command line tool (could be an npm package) that allows the flight controller to be put into MSC mode, and then use operating system level scripting, the aforementioned tool, the blackbox tools for data conversion, and plasmatree to do the task you describe.

simonbethke, 2019-12-28:
Well, the disintegration is what makes the process more tideous as it has to be. As a developer, you know to value an IDE. Keeping the bbl Explorer and the configurator development separated is one thing, but expecting the User to use and maintain two Tools hurts the ux more than just a bit.

For the sake of reducing the amount of duplicated code accross products, this tool belongs to the configurator. I don't understand why the msc mode can provide a better performance than any other simple binary transfer. Also afaik, the msc mode lacks of the possibility to clear the storage.

Although I know I do, I don't want to sound as if I knew stuff better than you do. I just think there are plenty of valid arguments to implement it not in just one more script-kiddy-project.

mikeller, 2019-12-29:
@simonbethke:

> For the sake of reducing the amount of duplicated code accross products, this tool belongs to the configurator. I don't understand why the msc mode can provide a better performance than any other simple binary transfer. Also afaik, the msc mode lacks of the possibility to clear the storage.

No, this does really not belong into configurator:
- file system manipulations - not configurator (and hard to do in a portable way in JavaScript to start with);
- log file conversion - not configurator (and there's already two projects that do this and need to be maintained);
- data analysis (two existing projects again).
All that is left is sending some MSP commands to the flight controller (to start MSC mode / delete the logs when all is finished). As I said, if we factor out the MSP communication into a library (which is something that I think will provide value beyond this), then taking care of this part can easily be done from any application.

As to why MSC as a native USB mode that was designed to allow performant file access would be better than siphoning megabytes or gigabytes of data through a control protocol that has a native limit of ~250 bytes payload per frame, and lacks the capability to process requests in parallel, limiting it to ~50 frames / second, ending up in maximum throughput of 40 kB/s with whatever optimisations are possible, I'll leave this to you to figure out.

Btw: MSC can delete log files on an SD card. It cannot delete 'log files' on on board storage, as the 'files' are just an emulation that we create on top of the raw log data that is stored onto the flash chip. Creating a file system on the flash chip does not make sense for the small size flash chips that a lot of the existing flight controllers have (we might think about it again for large size flash chips that are used by some newer flight controllers).

simonbethke, 2019-12-29:
Ok, I got it. So what about implenting an API in the configurator so that it can be used and partially controlled from outside?

mikeller, 2019-12-29:
@simonbethke:

> Ok, I got it. So what about implenting an API in the configurator so that it can be used and partially controlled from outside?

Yes, that would be one way to do it. Or, as I said, separate configurator into a 'worker' and a 'UI' part, and publish the worker as an npm module (or similar), so that it can be used by the UI _and_ by other applications - both of these ways should be workable.

closed 2020-02-04 by stale[bot]