betaflight/betaflight-configurator #1854: Error Installing 10.6

jyunte, 2020-01-04:
when I try to install betaflight-configurator-installer_10.6.0_win32.exe on my windows 10 system, I get the folowing error:

[image]

Not sure what I can do about this. I have an older version installed that runs fine (but does not support BF 4.1)

McGiverGim, 2020-01-04:
Your installation is corrupted for some unknown reason... 
The solution is to delete the old installation manually and install the new.
You must delete the folder of the old installation and the Betaflight registry keys. I will look later for them and publish them here if you are not able to search for them directly.

jyunte, 2020-01-04:
> Your installation is corrupted for some unknown reason...
> The solution is to delete the old installation manually and install the new.
> You must delete the folder of the old installation and the [...]

Thanks. I got several of those error messages for different files. I later did what you suggested (deleted the old installation and the folder), though I did not use the Windows Uninstall Program feature or remove the registry entries. It's interesting that I've never encountered this issue before when updating the Configurator. Anyway, I got it to successfully install.

jon2912, 2020-01-10:
Did you publish where the registry keys are? I have no clue how to find them

jon2912, 2020-01-10:
Do I delete the entire   C:\Program Files (x86)\Betaflight     folder?

McGiverGim, 2020-01-10:
Sorry, I didn't publish them because the other user fixed the problem. But the keys are:
HKLM\Software\Microsoft\Windows\CurrentVersion\Uninstall\Betaflight Configurator
HKCU\Software\Betaflight\Betaflight Configurator

McGiverGim, 2020-01-10:
And you must remove the Betaflight Configurator folder only, not the Betaflight folder that is parent of it (if you have the Blackbox installed it will be contained in this parent folder too).

jon2912, 2020-01-10:
I had no
HKLM\Software\Microsoft\Windows\CurrentVersion\Uninstall\Betaflight Configurator
I'm rebooting now.

jon2912, 2020-01-10:
I redownloaded  and attempted reinstallation. First one window displays

Betaflight Configurator Setup 
Installing 
Please wait while Betaflight Configurator is being installed.

Then another window partially coverning the first window displays

Betaflight Configurator Setup 
C:\Program Files(x86)\Betaf|ight\Betaf|ight-Configurator\uninbf00.dat is missing, some elements could not be removed. These can be removed manually.

jon2912, 2020-01-10:
[image]

McGiverGim, 2020-01-10:
The path to the uninbf00.dat that appears in your capture can be get from two places:
1. If you have administrative rights: HKLM\Software\Microsoft\Windows\CurrentVersion\Uninstall\Betaflight Configurator
2. If you don't have administrative rights: HKCU\Software\Betaflight\Betaflight Configurator

If you don't have the first, and you have deleted the second must be impossible for the installer search for the uninbf00.dat because it ignores where it is installed. So try to search for them. Maybe search for Betaflight Configurator across all the registry.

jyunte, 2020-01-10:
I uninstalled the Configurator by deleting the installation folder. I then
ran Priform's CCleaner product (free from Piriform's web site) using the
registry cleaner which removes unused registry entries. I then ran the
Configurator installer again, which completed successfully.

jon2912, 2020-01-11:
I successfully deleted betaflight and the two drivers (ST and SilLabs) and then reinstalled all. I can see them at
Control Panel\All Control Panel Items\Programs and Features
Still get message "Failed to open serial port".
When I assembled my fpv I was able to access the flight controller to calibrate the level and test the motor direction. Now I want to limit the amount of power to the motors but can't connect. I tried another pc (with fresh Win10 installation) and cable, no luck. The fpv was the only USB device. No lights come on the flight controller when I connect the USB cable. The drone does fly but being a novice (it's nothing like my Autel EVO is a gross understatement) I can't control it. I understand that I can limit throttle in the radio, but the flight controller can override those settings. Is it possible my flight controller became defective? It's Eachine GPS F4.

jyunte, 2020-01-11:
Try downloading, installing and running the ImpulseRC Driver Fixer, and
plug in the USB cable to your flight controller when told to do so.

jon2912, 2020-01-11:
The green slidy thingy above "Searching for flight controller" goes left to right again and again. No lights on fc. Nothing found.

closed 2020-02-17 by stale[bot]