betaflight/betaflight-configurator #1856: Betaflight USB vcp divers not working Beta flight configurator 10.0.6 Mac OS Catalina

qwasert-ben, 2020-01-10:
I can install beta flight and run it fine. I have installed the drivers for usb vcp drivers from link provided in beta flight.  Vcp drivers install fine, however, the com port does not show anything other than Bluetooth devices in drop down menu.  USB powers up and boots the board does not connect.

etracer65, 2020-01-11:
Try different USB cables. Many cables are "charge cables" and only supply power and don't have the data lines connected.

KK1203, 2020-01-19:
I have same symptom, running under catalina, cp210x installed, opens betaflight 10.6, com port just show BL devices.

When plug in FC, and click connect, in 10 secods it says "failed to open serial port".

KK1203, 2020-01-19:
Try all usb cables available at home, cannot all be only charging cables.

qwasert-ben, 2020-01-19:
I got mine to work. I downloaded 10.0.5 release and I only have one usb cable that works.  When I start betaflight I have to wait about 45 seconds then a window pops up that asks if I want to allow incoming connections and, accept. I plug in the quad and wait for the port to show up.

KK1203, 2020-01-19:
Exactly which version betaflight configurator ?

10.5.0 or 10.0.5 ?

KK

etracer65, 2020-01-19:
Do not try and use the 10.5 configurator with Betaflight 4.1 or later - it's not compatible and will corrupt settings. You must use 10.6. There is no problem with the configurator or difference in 10.5 that will affect connecting on a Mac. I just tested the 10.6 configurator on MacOS 10.15.2 and it worked perfectly.

Also you don't need to install any drivers on MacOS.

Lastly, go to the System Report and look at the USB option. If you don't see your flight controller there then your computer is not detecting the USB device and the Configurator cannot possibly connect to it. The problem most often is a bad USB cable.

KK1203, 2020-01-20:
I get attached screen.

KK

KK1203, 2020-01-20:
Attached screen with better resolution, it is Mac OS 10.5.2 Catalina, Mac book pro late 2012.

KK

etracer65, 2020-01-20:
You can’t attach images through email replies.

KK1203, 2020-01-20:
> [
>
> {
>
>
> "@type": "EmailMessage",
>
>
> "@type": "ViewAction",
>
> "target": "https://github.com/betaflight/betaflight-configurator/issues/1856?email_source=notifications\u0026email_token=AOKHSLW2PX53GZYNXPQWJS3Q6TYC7A5CNFSM4KFIBR42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJLBPDQ#issuecomment-576067470",
>
> "url": "https://github.com/betaflight/betaflight-configurator/issues/1856?email_source=notifications\u0026email_token=AOKHSLW2PX53GZYNXPQWJS3Q6TYC7A5CNFSM4KFIBR42YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJLBPDQ#issuecomment-576067470",
>
> "name": "View Issue"
>
> },
>
> "description": "View this Issue on GitHub",
>
> "publisher": {
>
> "@type": "Organization",
>
> "name": "GitHub",
>
> "url": "https://github.com"
>
> }
>
> }
>
> ]

I use Catalina 10.5.2, Betaflight 10.6.Tried with 2 MB, one 2012, one 2017, both same result, upper right display of betaflight configurator does not show serial port device, other than blue tooth devices. When connect to FC of quad, "fail to connect to serial port" message appears,  tried with several USB cables already.My FC requires betaflight_3.2.5OMNIBUSF4D.hexKKSent from my Samsung Galaxy smartphone.

KK1203, 2020-01-20:
Got it run now, you are right, I finally managed to get a USB cable that runs the configurator, and without any additional driver.

To reiterate, it is mac os Catalina 10.5.2, Betaflight configurator 10.6.

Thanks,

KK

qwasert-ben, 2020-01-20:
I have ordered a new USB data cable to connect the board, the only cable I have has a short in it. Beta Flight installation notes needs to include some info about using data cables not charge cables. I thought they were all the same. I see now this was probably my issue from the start.

etracer65, 2020-01-20:
@qwasert-ben Well you'll note that the first comment on the issue 9 days ago was:

> Try different USB cables. Many cables are "charge cables" and only supply power and don't have the data lines connected.

If this is resolved then please close the issue.

closed 2020-01-20 by qwasert-ben

uxinnuendo, 2022-08-03:
@etracer65 
I still see no mention on betaflight configurator "welcome" tips, per recommendation from @qwasert-ben to direct people to use a "data-transfer" cable, should this really have been closed.