betaflight/betaflight-configurator #1858: Is "using slot" in Adjustments missing

throttlenerd, 2020-01-19:
Hey guys,

just wanted my tinywhoop Rate Profiles to swap automatically with a flight mode change, watched a tutorial, but there is no "using slot" option between "then apply" and "via channel" in my Betaflight Configurator 10.6.0 on macOS 10.15.1. But an aged YouTube video clearly shows it used to be there )

Thanks!

McGiverGim, 2020-01-19:
It was removed because nobody understood the utility/functionality of the slots. I suppose you can do the same without them.

throttlenerd, 2020-01-19:
> It was removed because nobody understood the utility/functionality of the slots. I suppose you can do the same without them.

Ehmmm, as of now, I can't come up with an idea of how to set my AUX "low" to activate Rate Profile 1, "middle" -- 2 and "high" -- 3 rather than using an old slot method )

McGiverGim, 2020-01-20:
Simply assign a switch to the rate profile, without a slot, and it will work.

throttlenerd, 2020-01-22:
> Simply assign a switch to the rate profile, without a slot, and it will work.

Whew, really! Just made the AUX active throughout the whole 1000-2000 range and it switches 3 Rate Profiles back and forth! Thank you man )

closed 2020-01-22 by throttlenerd