betaflight/betaflight-configurator #1860: Video Transmitter Tab Blank

fOmey, 2020-01-22:
Title says it all, I can't see any fields on the video transmitter tab.. seems to have started happening after I had updated to 4.1.3? I _believe_ I was running 4.1.1 prior to this.

I immediately updated betaflight configurator thinking it would resolve the problem, however the issue persists... I couldn't tell you what my previous betaflight configurator version was tho.

I loaded in my VTX table txt file & saved it, I can see it was applied in the CLI.

[image]

McGiverGim, 2020-01-22:
As the message says, you must configure the kind of vtx you have in the ports tab.

fOmey, 2020-01-22:
I'm such a bloody doofus.. I don't know how I missed that.

Thank you.

closed 2020-01-22 by fOmey