betaflight/betaflight-configurator #1864: anything is wrong with serial config

TheIsotopes, 2020-01-26:
since latest configurator build the serial config have any issue

for example, i change the name of the quad and after comparing the diff this comes out:

left original and right after change the name
[image]

after looking through all the tabs, i also get a repeat of six readouts in the ports tab

[image]

I do not know if it is an configurator or a firmware issue

@fiam pls can you check this

mikeller, 2020-01-26:
Yop, got a reproduction:
- flash OMNIBUSF4 Unified Target with STM32F405 4.2.0 from `master` with configurator 10.7.0 from `master`;
- edit any setting in the Configuration tab;
- Save & Reboot.

Expected behaviour:
- the Ports tab shows 4 ports.

Actual behavior:
- the Ports tab shows 8 ports:

[image]

fiam, 2020-01-26:
Just found the problem for the duplicated ports. I’m working on the other problem right now and I’ll submit a PR in a bit.

mikeller, 2020-01-26:
First problem that I found:

```
mikeller@pc:~/git/betaflight-configurator$ git diff
diff --git a/src/js/msp/MSPHelper.js b/src/js/msp/MSPHelper.js
index e3d82d6d..a3380711 100644
--- a/src/js/msp/MSPHelper.js
+++ b/src/js/msp/MSPHelper.js
@@ -1896,7 +1896,6 @@ MspHelper.prototype.crunch = function(code) {
 
                 const functionMask = self.serialPortFunctionsToMask(serialPort.functions);
                 buffer.push32(functionMask)
-                    .push8(self.BAUD_RATES.indexOf(serialPort.msp_baudrate))
                     .push8(self.BAUD_RATES.indexOf(serialPort.msp_baudrate))
                     .push8(self.BAUD_RATES.indexOf(serialPort.gps_baudrate))
                     .push8(self.BAUD_RATES.indexOf(serialPort.telemetry_baudrate))
```

Not sure if it's the same that you found, but I'll hold off with more investigation until you let me know what you have.

fiam, 2020-01-26:
@mikeller That was the problem that was incorrectly setting the baudrates. I duplicated that line to test if the FC coped correctly with a payload longer than expected and then forgot to delete it.

The duplicated ports were caused because `SERIAL_CONFIG.ports` needed to be explicitly emptied, like in the handler for `MSP_CF_SERIAL_CONFIG`. I've just submitted fixed for both in #1865.

PR #1865 (2020-01-27): Fix problems introduced in #1851
- Clear the in-memory ports array when reading MSP2_COMMON_SERIAL_CONFIG,
like MSP_CF_SERIAL_CONFIG does. Take the opportunity to move the
`SERIAL_CONFIG.ports = [];` statement to the very first line of both
handler blocks, so it's more evident.
- Remove accidentally duplicated line in the payload serializer
for MSP_SET_CF_SERIAL_CONFIG, which then I copied and pasted to
MSP2_COMMON_SET_SERIAL_CONFIG. This was sending the msp_baudrate
twice, causing all baudrates after it to be misinterpreted.

Fixes #1864
commit 2b1c2e57 (2020-01-27): Fix problems introduced in #1851

- Clear to in-memory ports array when reading MSP2_COMMON_SERIAL_CONFIG,
like MSP_CF_SERIAL_CONFIG does. Take the opportunity to move the
`SERIAL_CONFIG.ports = [];` statement to the very first line of both
handler blocks, so it's more evident.
- Remove accidentally duplicated line in the payload serializer
for MSP_SET_CF_SERIAL_CONFIG, which then I copied and pasted to
MSP2_COMMON_SET_SERIAL_CONFIG. This was sending the msp_baudrate
twice, causing all baudrates after it to be misinterpreted.

Fixes #1864
closed 2020-01-27 by mikeller